openhwfoundation/cva6 #1: Move branch comparisons to ALU

zarubaf, 2018-02-07:
There are some potential area savings when moving the branch comparisons to the ALU. This will also ease timing on the write back path to the scoreboard/ROB.

raulbehl, 2018-03-05:
I have made the required changes and opened the #15 pull request. 
I was not able to verify whether the changes would lead to area savings as suggested in the bug description but I have verified the changes functionally. I have run the following tests with the changes in place: 
rv64ui-p-beq, rv64ui-v-beq
rv64ui-p-bne, rv64ui-v-bne
rv64ui-p-bge, rv64ui-v-bge
rv64ui-p-bgeu, rv64ui-v-bgeu
rv64ui-p-blt, rv64ui-v-blt
rv64ui-p-bltu, rv64ui-v-bltu

I did not see any fails with the above tests. Let me know if more validation is needed. 

Thank you!

zarubaf, 2018-03-06:
Again, really appreciate the contribution! Thanks a lot!

closed 2018-03-06 by zarubaf

raulbehl, 2018-03-06:
Thanks for the merge. Happy to contribute.