openhwfoundation/cva6 #2: Handle Bus Errors correctly

zarubaf, 2018-02-08:
At the moment bus errors are not resolved. Mostly because the AXI infrastructure does not support bus errors.

msfschaffner, 2019-02-11:
This could go hand in hand with PMA checkers at the cache refill ports.

MikeOpenHWGroup, 2023-02-18:
@zarubaf, was is the status of this enhancement? I recommend we either:
1. Set the `Status:Wont-fix` label and close it, or
2. Assign this issue to @JeanRochCoulon and ask for him to find an owner for this enhancement.

zarubaf, 2023-10-20:
We should fix this really :-)

niwis, 2023-10-20:
In case it's relevant to this discussion, @micprog has developed a bus error unit to track asynchronous bus errors: https://github.com/pulp-platform/unbent
It sends an interrupt to the core if a bus error occurs and stores some additional information for handling the bus error by an interrupt handler.