openhwfoundation/cva6 #4: Make verilator

lfernand4, 2018-02-14:
Hi, I am getting the following issue while running "make verilate" command:
[image]

zarubaf, 2018-02-16:
This is because your `riscv-fesvr` installation is not in your compiler's/linker's path. Also note that for the moment Ariane requires our forked version of `riscv-fesvr` (which can be found  [here](https://github.com/pulp-platform/riscv-fesvr)).

lfernand4, 2018-02-16:
I install "riscv-fesvr" in an other directory. How can I add it to my Path ?

zarubaf, 2018-02-16:
As with any other library you don't have in your default path. 
```
export LIBRARY_PATH=$LIBRARY_PATH:/fesvr/installation/lib
export LD_LIBRARY_PATH=$LIBRARY_PATH:/fesvr/installation/lib
export C_INCLUDE_PATH=$LIBRARY_PATH:/fesvr/installation/include
export CPLUS_INCLUDE_PATH=$LIBRARY_PATH:/fesvr/installation/include
```
You might want to put this into your shell's start-up script.

lfernand4, 2018-02-16:
Thank you for your help.
I have a new error whith the make verilate : 
[image]

zarubaf, 2018-02-16:
It seems you are using the official version and not our fork, which I linked in the previous message. 

> Also note that for the moment Ariane requires our forked version of `riscv-fesvr` (which can be found  [here](https://github.com/pulp-platform/riscv-fesvr)). 

You can use our version with the rest of the RISC-V ecosystem though. The changes are the visibility of the two variables (which is the error you get ;-))

lfernand4, 2018-02-16:
This Path is for the fork : /prj/xpuips/riscv/riscv-fesvr/include
We can see that it use the good path.

lfernand4, 2018-02-16:
I think that the path is good. 
/prj/xpuips/riscv/riscv-fesvr/ is the good path.

zarubaf, 2018-02-16:
Yes, you are right - thanks for reporting. The changes were on a branch (`expose-htif`). Can you try to checkout the newest changes from `pulp-platform/riscv-fesvr` and rebuild it?

lfernand4, 2018-02-16:
It work ! Nice

lfernand4, 2018-02-19:
New problem : 
[image]
It is because my riscv-fesvr installation is not in my compiler's/linker's path ?

lfernand4, 2018-02-19:
Now I want to run dhrystone. I have the riscv-tests. But i don’t know how to configure the good ISA.

This work : 
Variane_wrapped -p /prj/xpuips/riscv/riscv-tests/benchmarks/dhrystone.riscv

Can you help me ?

zarubaf, 2018-02-19:
So theoretically this should work. Takes a couple of seconds though ;-) Are you experiencing any issues?

Also note that there is a branch on which I am working right now which significantly improves the IPC (Drystone ~1.71).

lfernand4, 2018-02-19:
The problem is I found 1.19. I think i'am making a mistake. I don't configure the ISA.

zarubaf, 2018-02-19:
Im afraid I can't really follow. Everything should be compatible to the latest RISC-V compiler and RISCV-tests (at least one week ago ;-)). You can't really tweak the uArch but only the compiler.

lfernand4, 2018-02-19:
ok

lfernand4, 2018-02-26:
In the makefile from : riscv-tests/benchmarks. I don't found where the -march is configure for the compiler. That's why I think the drystone results are incoherent.

zarubaf, 2018-02-28:
Yes the benchmarks get compiled with the full compiler functionality. But nevertheless, Dhrystone doesn't use either floating points nor atomics. So it should work out of the box.

lfernand4, 2018-02-28:
When I run dhrystone out of the box I found : 
[image]
It is correct ? and how to convert in Dhrystones/MHz ? (500 runs and Hz = 1.000.000)
It is : (HZ * Number_Of_Runs) / (1757*mcycle) ?

zarubaf, 2018-02-28:
That looks like the supposed outcome. That being said it seems that the Dhrystone report can't be taken as an absolute value ([see this thread](https://groups.google.com/a/groups.riscv.org/forum/#!topic/sw-dev/b2eTdNOtjhA)). You can use it though to compare to Rocket for example.

lfernand4, 2018-02-28:
What do you use to run Dhrystone ? riscv-tests ?

zarubaf, 2018-03-02:
At the moment yes, just for relative comparisons though.

Nicu1309, 2018-04-13:
Hello, I am trying to run the "make verilate" command, but I am getting several syntax errors. I guess this could be related to the verilator version that I am using (Verilator 3.874). Could you please confirm?

[image]

Thanks :)

Nicu1309, 2018-04-17:
Hey @raulbehl, 
perfect, I got it built with Verilator 9.20. Thanks a lot 😃

closed 2018-08-23 by zarubaf