openhwfoundation/cva6 #6: Q: Code & Functional Coverage?

ben-marshall, 2018-02-17:
Hi All

This is a question rather than an 'issue' - apologies if I am asking it in the wrong place, I'm happy to close this and ask it again there.

How do you collect code coverage and functional coverage numbers from your verification environment? I'm working on a similar project and am stuck in terms of what tools support. What you used to verify against (as a reference model) and check you were covering different conditions the tests say they should?

Thanks,
Ben

zarubaf, 2018-02-19:
Hi Ben,

so I am not sure I understand your question a 100%, please ask again if it needs further clarification.

So code coverage (in terms of covered SV lines) is done with ModelSim (QuestaSim) which provides all the necessary functionality to do so (you can have a look at the Makefiles if you want). CI is also able to extract the coverage metric by parsing the log (see the .gitlab.ci as a ref).

That being said it is certainly quite hard to get a reasonable confidence that the tests are covering important paths. A good start are certainly the `riscv-tests` as well as the torture test-framework. But both are far from being perfect. Especially what concerns compressed instructions.

Best,

Florian

ben-marshall, 2018-02-20:
Hi Florian

Yeah sorry, I should have been clearer. I understand that code coverage (like you say, SV lines) can be done with ModelSim, or indeed any other sim.

My question is about *functional* coverage. That is, covering which pieces of RISC-V functionality have been implemented and stressed. For example, how do you know your constrained random testing setup generates the instruction sequence "X;Y;Z", and that the core also executed that sequence?

I certainly agree that the riscv-tests are a good sanity check. I've used them as a sort of "first spin" milestone in other projects, but you're right in saying that both they and torture are far from comprehensive. Yours is one of the larger open source RISC-V cores I have seen, and I was curious about how you handled  verification tooling. I'm restricted at the moment to open / free tools (Icarus, Verilator etc) and so gathering any sort of coverage can be very difficult.

Thanks,
Ben

zarubaf, 2018-02-20:
Hi Ben,

okay I thought so. So you are absolutely right with your observation that it is really hard to gain a reasonable amount of verification, especially with less stressed paths (like address translation and interrupts, etc). So what I did in those regards:

1. I've implemented a couple of unit-test testbenches (ALU, instruction fetch, etc.) in a randomly constrained verification environment.
2. `riscv-tests` and `riscv-torture`
3. What ultimately uncovered quite a lot of bugs was booting Linux on the FPGA/RTL simulation.

That said, it is far from being perfect but as much as I could get running in a single person effort. I am not a verification expert but what in the next months will be on the plate regarding verification is to get `riscv-pk` running (as soon as atomics are properly implemented) is to get small SW tests running (like GCCs torture tests).

Hope that helps.
Best,
Florian