openhwfoundation/cva6 #7: Generate hex file

hoangt, 2018-02-25:
What is the command to generate .hex from .elf for Questasim?

I used the following command but not sure that is correct (just run few ISA tests and they worked fine but when I run rsort benchmark, it is run out memory)

elf2hex 8 16384 hello.riscv 2147483648 > hello.riscv.hex

zarubaf, 2018-02-25:
The `elf2hex` command is absolutely correct. The reason why you run out of memory is that Questasim is not using `riscv-fesvr` which handles the communication with the target. It just implements the the end-of-computation signal coming from the DUT (store to `.tohost` address with LSB set). This works fine with the basic `riscv-tests` but fails on the benchmarking suit as all benchmarks want to write to stdout.  The benchmark is waiting for an acknowledgement of the host which it is basically never getting.

That beeing said, I am unsure whether I will implement the missing functionality because the way host-target communication is currently handled within `riscv-fesvr` and `riscv-tests` is quite messy with non-cache coherent systems. The Verilator testbench handles this simulation though and should let you do the benchmarking.

zarubaf, 2018-02-28:
Would be happy to accept a PR with this added functionality. Yes, indeed - we do have a SoC based on this core which also supports FPGA mapping. It is still in somehow a messy state but will be pushed quite soon.

closed 2018-08-23 by zarubaf