openhwfoundation/cva6 #10: Questasim uvm_dpi.so

lfernand4, 2018-02-28:
HI,

I have build the TB and module with questasim 10.6c_3.  I start the simulation : make simc (without GUI)
[image]

 I have the following warning : 
[image]

Now I run (run -a) : 
[image]

How can I solve this error ? where does it come from ?
I have only add -64 in the scipt.

Thanks in advance.

raulbehl, 2018-03-01:
Even I am seeing the same issue. I do see in the Makefile that the C-code is compiled using the vlog command only:
```
# build TBs
$(library)/.build-tb: $(dpi) $(tbs)
	# Compile top level with DPI headers
	vlog$(questa_version) -sv $(tbs) -work $(library) $(filter %.c %.cc, $(dpi)) -ccflags "-g -std=c++11 " -dpiheader tb/dpi/elfdpi.h
	touch $(library)/.build-tb
```

Questasim should have been able to load the compiled .so files automatically but for some reason it is not doing that. 
As a work around to get things working, I did the following:
- Compiled the tb/dpi/*.c files separately:
```
g++ -c -fPIC -m64 -std=c++0x -I$QUESTASIM_HOME/include -shared -o elf_dpi.o -c elfdpi.cc
g++ -shared -m64 -o elf_dpi.so elf_dpi.o
```
- Updated the simc target of the Makefile
```
vsim${questa_version} -64 -c -lib ${library} ${top_level}_optimized +max-cycles=$(max_cycles) +UVM_TESTNAME=${test_case} \
	 +BASEDIR=$(riscv-test-dir) $(uvm-flags) +ASMTEST=$(riscv-test) -coverage -classdebug -sv_lib ariane_core/ariane/tb/dpi/elf_dpi -do "do tb/wave/wave_core.do"
```
With the above hack I was able to get it working. The correct fix should enable questasim to included those libraries without doing a separate g++ compile.

lfernand4, 2018-03-02:
Thanks for explanations. Now I want to run dhrystone, do you know how to do it ? 
I have compiled in .elf (.riscv). I feel I need a .hex. It is correct ?

zarubaf, 2018-03-02:
Yes, I've added some documentation on that in the README.md`.
`elf2hex 8 2097152 elf_file.riscv 2147483648  > elf_file.riscv.hex`

zarubaf, 2018-03-02:
@raulbehl Thanks for clarifying the issue with DPI. I think I will investigate a better approach to that, considering something like you did (not using Questa for it).

raulbehl, 2018-03-02:
@zarubaf thank you for the comment. Along with the DPI change, the vsim command in the Makefile also needs the "-64" option passed to it. Otherwise questa complains that the code was compiled with 64-bit support but is not simulated with it.

Also, would it be okay if I add the above mentioned changes to handle DPU calls in the Makefile and open a pull request for the same? You could then add your fixes on top of  it assuming those are on the similar lines.

lfernand4, 2018-03-02:
I have generate Dhrystone.riscv.elf and Dhrystone.riscv.hex. 

command : 
****

I have this error in questasim : 

[image]

How can I solve this error ? where does it come from ?

Thanks in advance.

zarubaf, 2018-03-02:
@lfernand4 See issue #7
@raulbehl I would be more than happy to accept such a PR. Thanks!

closed 2018-03-05 by zarubaf