openhwfoundation/cva6 #11: error in "make build"

merchantf, 2018-03-01:
Hi, 
I am getting following error when I do "make build"

# Optimize top level
vopt +cover=bcfst+/dut -incr -64 -nologo -quiet -suppress 13262 -permissive -work work  core_tb -o core_tb_optimized +acc -check_synthesis
** Error: tb/core_tb.sv(132): Expected an interface instance as an actual for 'master'.
** Error: tb/core_tb.sv(132): Expected an interface instance as an actual for 'slave'.
Optimization failed
make: *** [build] Error 2

Can you help me with this error?

Thanks  in advance. 

Farhad

zarubaf, 2018-03-02:
Can you post the whole build log? Are you using the master branch?

merchantf, 2018-03-02:
[make_build_log.txt](https://github.com/pulp-platform/ariane/files/1774662/make_build_log.txt)
[make_verilate_log.txt](https://github.com/pulp-platform/ariane/files/1774663/make_verilate_log.txt)

Yes. I am using master branch. 

I am attaching here make_verilate_log.txt and make_build_log.txt

zarubaf, 2018-03-02:
That should be fine. Which version of Questa are you using?

farhadmerchant, 2018-03-02:
I have several versions of modelsim. 

From Makefile, I have to remove version information to run "make build"

because whichever version of Modelsim I load, it give be error that vlib-10.6 command not found. 

So, I remove version from the Makefile and then run the command. 

Is that correct way to do it?

merchantf, 2018-03-05:
Hi,

This is a gentle reminder. If you could help me to solve this problem. 

Thanks & Regards,

Farhad

zarubaf, 2018-03-05:
@farhadmerchant Yes, the version flag is specific for the setup in ETH, so in the general case you won't need to specify it.
@merchantf It is really difficult to say what causes the problem at your end. Are you sure you are using a full licensed Questa version? It could be that (for example the student version) doesn't support certain features.

merchantf, 2018-03-05:
[make_build_log_new.txt](https://github.com/pulp-platform/ariane/files/1780913/make_build_log_new.txt)
I changed the version and I have attached the log file. "make build" ends with "notes" on FSM, I am not sure if it is correct. 

Going further from here, I ran this command 

$ elf2hex 8 2097152 elf_file.riscv 2147483648  > elf_file.riscv.hex
elf2hex: ../fesvr/elfloader.cc:21: std::map<std::basic_string<char>, long unsigned int> load_elf(const char*, memif_t*, reg_t*): Assertion `fd != -1' failed.

For elf2hex, I have sourced "bin" directory in my .bashrc from riscv-tools 

Is this the correct way to do it?

zarubaf, 2018-03-05:
Yes that is the correct way and also the build seems to be fine. 
```
Assertion `fd != -1' failed.
```
Tells you that the input file is not a valid file descriptor. Not a problem specific to Ariane though ;-)

merchantf, 2018-03-05:
I tried running "make simc" and it failed with this error. 

# Loading mtiUvm.questa_uvm_pkg(fast)
# Loading /tmp/merchantf@tetanus_dpi_29075/linux_x86_64_gcc-4.7.4/vsim_auto_compile.so
# Loading /net/home/merchantf/Downloads/ModelSim-installation/mentor/2016-17/RHELx86/QUESTA-CORE-PRIME_10.5c-4/questasim/uvm-1.1d/linux_x86_64/uvm_dpi.so
# do tb/wave/wave_core.do
VSIM 2> run
vsimk: tb/dpi/elfdpi.cc:61: void* read_elf(const char*): Assertion `fd != -1' failed.
# ----------------------------------------------------------------
# UVM-1.1d
# (C) 2007-2013 Mentor Graphics Corporation
# (C) 2007-2013 Cadence Design Systems, Inc.
# (C) 2006-2013 Synopsys, Inc.
# (C) 2011-2013 Cypress Semiconductor Corp.
# ----------------------------------------------------------------
# UVM_INFO verilog_src/questa_uvm_pkg-1.2/src/questa_uvm_pkg.sv(215) @ 0: reporter [Questa UVM] QUESTA_UVM-1.2.3
# UVM_INFO verilog_src/questa_uvm_pkg-1.2/src/questa_uvm_pkg.sv(217) @ 0: reporter [Questa UVM]  questa_uvm::init(+struct)
# UVM_INFO @ 0: reporter [Program Loader] Pre-loading memory from file: tmp/riscv-tests/build/isa/rv64ui-p-add
# 
# ** Warning: (vsim-7) Failed to open readmem file "tmp/riscv-tests/build/isa/rv64ui-p-add.hex" in read mode.
# No such file or directory. (errno = ENOENT)    : tb/env/core/core_test_util.svh(55)
#    Time: 0 ps  Iteration: 10  Instance: /core_tb/tb
# ** Fatal: (SIGABRT) Bad handle or reference.
#    Time: 0 ps  Iteration: 10  Process: /core_tb/tb/#INITIAL#248 File: tb/core_tb.sv
# Fatal error in Module core_tb at tb/core_tb.sv line 29
# 
# HDL call sequence:
# Stopped at tb/core_tb.sv 29 Module core_tb
# called from  tb/core_tb.sv 255 Module core_tb

Any idea why is there a "fatal error"?

zarubaf, 2018-03-05:
Yes, because:
```
Warning: (vsim-7) Failed to open readmem file "tmp/riscv-tests/build/isa/rv64ui-p-add.hex
```
Pls, have a look at the `README.md` how to run arbitrary programs. By default it fetches the programs from that path (because of internal CI reasons). Also make sure that the elf and `.hex` are in the same location.

merchantf, 2018-03-05:
just a dumb question. Where do I get elf file(s)? Is it inside riscv-tools or is it inside airane directory?

For example, in this command 
obj_dir/Variane_wrapped -p rv64um-v-divuw

where do I get rv64um-v-divuw file? Are these the files inside $RISCV/bin ?

raulbehl, 2018-03-05:
Hi @merchantf, 
I doubt if you would find the ELFs already present. I think you would have to generate the ELFs using the RISCV-tools. This is what I did to run my sims using verilator/questasim. You can find the riscv-tools repo at [riscv-tools](https://github.com/riscv/riscv-tools)

merchantf, 2018-03-05:
Hi,

I have riscv-tools already built in my system. is it inside riscv-tools/bin directory?

raulbehl, 2018-03-05:
If you already have the riscv-tools installed, you should be able to generate the tests using [riscv-tests](https://github.com/riscv/riscv-tests)

Do a make inside the `isa` folder to get the ELFs generated.

merchantf, 2018-03-05:
Got it..I have several .dump files  and executable files inside riscv-tests.

closed 2018-08-23 by zarubaf