openhwfoundation/cva6 #16: ISA Tests

merchantf, 2018-03-07:
I ran ISA test with elf file "rv64ud-v-fadd". 

Output I see when I run following command
make simc riscv-test=rv64ud-v-fadd

# do tb/wave/wave_core.do
VSIM 2> run
# ----------------------------------------------------------------
# UVM-1.1d
# (C) 2007-2013 Mentor Graphics Corporation
# (C) 2007-2013 Cadence Design Systems, Inc.
# (C) 2006-2013 Synopsys, Inc.
# (C) 2011-2013 Cypress Semiconductor Corp.
# ----------------------------------------------------------------
# UVM_INFO verilog_src/questa_uvm_pkg-1.2/src/questa_uvm_pkg.sv(215) @ 0: reporter [Questa UVM] QUESTA_UVM-1.2.3
# UVM_INFO verilog_src/questa_uvm_pkg-1.2/src/questa_uvm_pkg.sv(217) @ 0: reporter [Questa UVM]  questa_uvm::init(+struct)
# UVM_INFO @ 0: reporter [Program Loader] Pre-loading memory from file: /net/home/merchantf/Downloads/riscv-tools/riscv-tests/build/isa/rv64ud-v-fadd
# 
# UVM_INFO @ 0: reporter [Program Loader] tohost: 0000000080001000 begin_signature 0000000080003140
# 
# UVM_INFO @ 0: reporter [RNTST] Running test core_test...

Is this expected output?

raulbehl, 2018-03-07:
Hi @merchantf, 

Please use `run -a` instead of just run. `run` would just do the things supposed to happen at t=0ns.

zarubaf, 2018-03-07:
Thanks for clarifying @raulbehl 
Just as a note: The core should ideally fail at executing that test if you have a hard-float API enabled.

raulbehl, 2018-03-07:
@zarubaf thank you for the information on hard-float API. 

Looking at this bug, I just realized if it would be better to add the `run -a` as part of the simc target itself in the Makefile?

zarubaf, 2018-03-07:
Yes actually a very good idea!

zarubaf, 2018-03-08:
I think we can safely close this issue now. Nobody should run into that condition anymore.

closed 2018-03-08 by zarubaf