openhwfoundation/cva6 #18: Instruction trace when using Verilator

raulbehl, 2018-03-08:
Hi, 

When using Verilator for running ELFs, I see that only the performance numbers are printed on stdout. Apart from that we can dump the signals to create a .vcd file. Wanted to know if there is an option to dump the executed instruction trace as well (similar to the way we see on Questasim)? That would make debugging easier when working with Verilator. 

Thank you!

zarubaf, 2018-03-08:
Probably something which I should add to the README.

raulbehl, 2018-03-08:
Great. Thank you for the information. This should be helpful for now.

ShravanSubbaraman, 2018-07-18:
You can run the verilator command using the following command to directly get the vcd file pop up in the screen : LD_LIBRARY_PATH=/home/riscv/lib/ obj_dir/Variane_wrapped --vcd=xyz.vcd -p xyz

closed 2018-08-23 by zarubaf