openhwfoundation/cva6 #19: Memory Failure

merchantf, 2018-03-08:
# UVM_INFO @ 0: reporter [Program Loader] tohost: 0000000080001000 begin_signature 0000000080003000
# 
# UVM_INFO @ 0: reporter [RNTST] Running test core_test...
# [TRACER] Output filename is: trace_core_00_0.log
# UVM_INFO tb/env/core/core_eoc.svh(59) @ 371570000: uvm_test_top.m_env.m_eoc [Core Test] *** SUCCESS *** (tohost = 0)
# UVM_INFO verilog_src/uvm-1.1d/src/base/uvm_objection.svh(1268) @ 371570000: reporter [TEST_DONE] 'run' phase is ready to proceed to the 'extract' phase
# ** Fatal: (vsim-5) ****** Memory failure. *****
# Bad pointer/access type passed to memory subsystem.
# Pointer may have been previously deallocated.
# ** Fatal: (vsim-5) ****** Memory failure. *****
# Bad pointer/access type passed to memory subsystem.
# Pointer may have been previously deallocated.
# 
# --- UVM Report Summary ---
# 
# ** Report counts by severity
# UVM_INFO :    7
# UVM_WARNING :    0
# UVM_ERROR :    0
# UVM_FATAL :    0
# ** Report counts by id
# [Core Test]     1
# [Program Loader]     2
# [Questa UVM]     2
# [RNTST]     1
# [TEST_DONE]     1
# ** Note: $finish    : /net/home/merchantf/Downloads/ModelSim-installation/mentor/2016-17/RHELx86/QUESTA-CORE-PRIME_10.5c-4/questasim/linux_x86_64/../verilog_src/uvm-1.1d/src/base/uvm_root.svh(430)
#    Time: 371570 ns  Iteration: 50  Instance: /core_tb/tb
# End time: 11:24:02 on Mar 08,2018, Elapsed time: 0:00:07
# Errors: 2, Warnings: 0

Any idea why do I get into memory failure for "rv64um-v-divu"?

zarubaf, 2018-03-09:
First of all the test is passing: `uvm_test_top.m_env.m_eoc [Core Test] *** SUCCESS *** (tohost = 0)`
Until now I haven't found the issue why Questa is complaining. Probably I am referencing some de-allocated poitner. Sorry for that. Nevertheless, simulation is valid and passing ;-)

raulbehl, 2018-03-12:
When you do `verror 5`, the help usually points to a de-allocated object being used in a `$fwrite/$display` function. I tried removing a couple of such writes present in the phases after the run_phase (since run_phase exits smoothly) but with little luck. Will continue looking at it and update if I find anything.

mansijain1234, 2018-06-01:
Hi,

I am facing an issue in the test getting Success. I am getting it as failed. 

[image]

zarubaf, 2018-07-18:
This will be fixed by #3

closed 2018-08-23 by zarubaf