openhwfoundation/cva6 #20: Build error on VCS

kuopinghsu, 2018-03-09:
I got a problem to build the ariane RTL code on VCS. It happens on the miss_handler.sv line #684,

#683:    // counter for AXI transfers
#684:    logic [$clog2(DATA_WIDTH/64)-1:0] cnt_d, cnt_q;
#685:    logic [(DATA_WIDTH/64)-1:0][63:0] cache_line_d, cache_line_q;

Here is the patch,
#684:    logic [ADDR_INDEX-1:0] cnt_d, cnt_q;

zarubaf, 2018-03-09:
Thanks for the patch. I will take care of it. What is the problem VCS is giving?

kuopinghsu, 2018-03-12:
If the DATA_WIDTH is 64, then the $clog2(DATA_WIDTH)/64 - 1 will be negative value. The error message of VCS is,

Error-[UCTLAB] Unsigned constant too large
src/miss_handler.sv, 684
"($clog2((DATA_WIDTH / 64)) - 1)"
  Unsigned compile-time constant used as left dimension bound is greater than
  2147483647.
  VCS does not support values greater than 2147483647 as array bound.

raulbehl, 2018-03-12:
If that is the case then it should cause some functional issues in questa as well (assuming it doesn't give any compile time errors).

closed 2018-06-20 by zarubaf