openhwfoundation/cva6 #21: Error in elf2hex

merchantf, 2018-03-09:
Hi,

I am getting following error in generating hex file 

elf2hex 8 2097152 /net/home/merchantf/Downloads/riscv-tools/hello 2147483648  > hello.hex
elf2hex: ../fesvr/htif_hexwriter.cc:17: virtual void htif_hexwriter_t::read_chunk(addr_t, size_t, void*): Assertion `taddr < width*depth' failed.

Can you help me with this?

zarubaf, 2018-03-09:
I am assuming your baseaddress is not correct and therefore the hexwriter generates an overflow. So summing it up this can have two causes:

1. Your linker script does not place the instructions at `0x80000000` (base is the last parameter `d2147483648` of the call).
2. The program you have written overflows the size of the memory (the second parameter of `elf2hex`), that is unlikely.

Here is the usage of `elf2hex`:

```
elf2hex --help                                                     
Usage: elf2hex <width> <depth> <elf_file> [base]
```

merchantf, 2018-03-09:
My program is same hello program given in riscv-tools github page. 

I could generate file but it seems hello.hex file has some problem. 

When i run command 

make simc riscv-test=hello

I get this error:

vsimk: tb/dpi/elfdpi.cc:61: void* read_elf(const char*): Assertion `fd != -1' failed.
# ----------------------------------------------------------------
# UVM-1.1d
# (C) 2007-2013 Mentor Graphics Corporation
# (C) 2007-2013 Cadence Design Systems, Inc.
# (C) 2006-2013 Synopsys, Inc.
# (C) 2011-2013 Cypress Semiconductor Corp.
# ----------------------------------------------------------------
# UVM_INFO verilog_src/questa_uvm_pkg-1.2/src/questa_uvm_pkg.sv(215) @ 0: reporter [Questa UVM] QUESTA_UVM-1.2.3
# UVM_INFO verilog_src/questa_uvm_pkg-1.2/src/questa_uvm_pkg.sv(217) @ 0: reporter [Questa UVM]  questa_uvm::init(+struct)
# UVM_INFO @ 0: reporter [Program Loader] Pre-loading memory from file: /net/home/merchantf/Downloads/riscv-tools/riscv-tests/build/isa/hello
# 
# ** Fatal: (SIGABRT) Bad handle or reference.
#    Time: 0 ps  Iteration: 10  Process: /core_tb/tb/#INITIAL#248 File: tb/core_tb.sv
# Fatal error in Module core_tb at tb/core_tb.sv line 29
# 
# HDL call sequence:
# Stopped at tb/core_tb.sv 29 Module core_tb
# called from  tb/core_tb.sv 255 Module core_tb

I am not able to attach hex file here but, I am not sure if it is correctly generated hexfile since initiall 8191 values seem to be zero

zarubaf, 2018-03-09:
A hex file wouldn't help me anyway. Can't read hex that fluently :)
The testbench can't find your elf file. Make sure it is there.
`void* read_elf(const char*): Assertion `fd != -1' failed.`
Both the hexfile as well as the elf file need to be there. Because the testbench is extracting some symbol addresses from the elf. Maybe in a future version it will only require the elf.

merchantf, 2018-03-09:
I placed elf file in the same directory as hex file. 

Now it runs but goes into infinite simulation it seems. I have a doubt on my elf2hex command. 

elf2hex 8 2097152 /net/home/merchantf/Downloads/riscv-tools/hello 16 > hello.hex

Is this correct way of using elf2hex?

zarubaf, 2018-03-09:
That is something I can't answer in general. A potential problem I see here is that you've changed the base address. If you didn't change anything in the RTL this will break (Ariane by default is fetching from `0x80000000`).

merchantf, 2018-03-09:
If I run this command

elf2hex 8 2097152 /net/home/merchantf/Downloads/riscv-tools/hello 2147483648  > hello.hex

which is same as given on ariane github page. it gives me following error. 

elf2hex: ../fesvr/htif_hexwriter.cc:17: virtual void htif_hexwriter_t::read_chunk(addr_t, size_t, void*): Assertion `taddr < width*depth' failed.

But the same command works for other isa tests. 

Any idea why this happens?

zarubaf, 2018-03-09:
I would bet that your linker script might be wrong. Can you post the first couple of lines of the following output:
`riscv64-unknown-elf-objdump -D path/to/elf `

merchantf, 2018-03-09:
[log.txt](https://github.com/pulp-platform/ariane/files/1797754/log.txt)
Here it is...

zarubaf, 2018-03-09:
That's what I guessed: You are placing instructions at `0x...00`. That won't work by default. In general I think you are confusing some stuff here. You are trying to run a program which clearly requires a POSIX API bare metal. That won't work.

merchantf, 2018-03-09:
My program is simple 

#include<stdio.h>
int main(){
printf("Hello World");
return 0;
}

How do I run above program?

zarubaf, 2018-03-09:
Your best shot is to use the newlib compiler and link it against the `riscv-tests` API which provides a `printf` which is actually just opening a write to stdout (see the `vadd` or so). But you will need to use the Verilator model for that because the program needs the front-end server for communication with host.

merchantf, 2018-03-09:
Ok. Thanks. 

Is it possible to get step-by-step clear instructions for that?

raulbehl, 2018-03-12:
Hi @merchantf, 

I was able to run the hello.c program on the Verilator simulator as suggested. Here are the steps I followed: 

Compile the .c file using the following command (you need to have the riscv-tests repo checked-out):
```
riscv64-unknown-elf-gcc -I./riscv-tests/benchmarks/../env -I./riscv-tests/benchmarks/common \
-DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O2 -ffast-math -fno-common \
-fno-builtin-printf -o hello.riscv hello.c ./riscv-tests/benchmarks/common/syscalls.c \
./riscv-tests/benchmarks/common/crt.S -static -nostdlib -nostartfiles -lm -lgcc \
-T ./riscv-tests/benchmarks/common/test.ld
```
Use the generated ELF file as an input to the Verilator model:
```
obj_dir/Variane_wrapped -p --vcd=dump.vcd hello.riscv
```
[image]

merchantf, 2018-03-13:
It is working fine. Thanks a lot. 

What do cycles indicate here? I am getting 4450+ cycles.

zarubaf, 2018-03-13:
Thanks a lot @raulbehl ! Should we add this to a separate README section? Furthermore I actually want to try for quite some time to use `riscv-pk`. Actually this could work out of the box if we compile `riscv-pk` with a compiler which does not support the A extension. Basically the Verilator model should behave exactly as Spike `--isa=rv64imc` (excluding timer support).

Cycles indicate how many cycles your DUT did since the start. Its more an indication on how fast the C++ model simulates. If you don't launch the model with the `-p` flag it will not print anything of that information. (p stands for performance counters)

raulbehl, 2018-03-13:
@zarubaf, I think it would be good to add this in the README section. It would help anyone who would want to run their own C code/benchmarks on Ariane. 

We could definitely try out the `riscv-pk` stuff. It would be interesting to see how it goes. At this point of time I am not sure how can we ensure that the atomics are not generated by the compiler but there should be some configurable option available (might involve building the riscv-gcc from the source directly).  
Though I see that you already have the `a-extensions` branch in the repo. I think it would be better if we add that support in Ariane and then try out `riscv-pk` ? It would be great if I could help out as well. What do you suggest?

zarubaf, 2018-03-14:
Thanks for the PR. I am currently trying to fix the CI build as it is failing for some dubious reason and merging afterwards. Regarding the atomic extension I will open a new issue describing the current situation and needed implementation. Any help is greatly appreciated although I think that is quite a non-trivial addition.

raulbehl, 2018-03-14:
Yes, I understand that the implementation might not be straightforward and I might end up asking a lot of questions/continuously bothering you for help. But as you said, it would be a good way to get an in-depth understanding of the design. :) 
Thanks again!

closed 2018-03-14 by zarubaf

raulbehl, 2018-03-14:
Thank you for merging the Pull request. Like you said, working on a-extension stuff might give me time to ramp up on the design and understand it better. :)

zarubaf, 2018-03-14:
Currently up to the task of summarizing what will be needed (imho) to support it. Bare with me a little :-)

raulbehl, 2018-03-14:
Sure. Thank you so much :-)

commit 96eb8cdf (2018-03-14): Merge pull request #22 from raulbehl/master

README update for (fix #21)