openhwfoundation/cva6 #25: A-Extension Support (Atomic Memory Operations)

zarubaf, 2018-03-14:
This issue targets the long needed atomic memory support. 

## Overview

General there are 4 cases:

**acquire semantic**: If the `aq` bit is set, then no later memory operations in this RISC-V hart can be observed to take place before the AMO.
-> block issue until the atomic memory operation is resolved, wait on commit signal, flush cache, do AMO, write back

**release semantic**: If the `rl` bit is set, then other RISC-V harts will not observe the AMO before memory accesses preceding the AMO in this RISC-V hart.
-> block issue until no more outstanding loads (trivially met as the load unit will implicitly synchronize, structural hazard) or stores (wait for store buffer to drain), wait on AMO commit, flush cache, do AMO, write-back

**sequentially consistent**: If both the `aq` and `rl` bits are set, the atomic memory operation is sequentially consistent and cannot be observed to happen before any earlier memory operations or after any later memory operations in the same RISC-V hart, and can only be observed by any other hart in the same global order of all sequentially consistent atomic memory operations to the same address domain.
-> block issue until no more outstanding loads or stores, issue atomic, wait on commit signal, block issue until atomic comited, flush dcache, do amo, write-back

**unordered**: aq and rl not set -> just issue -> wait for amo_commit -> flush dcache -> do AMO -> write back AMO

## Details

- **Decoder** (`decoder.sv`): The atomic instruction needs to be properly decoded. Most of the operations are there but this needs some further checking (especially if all parameters are set properly and illegal instructions are thrown in case of an illegal instruction with the `OPCODE_AMO`).
Decoder will also need to block the decoding of any subsequent instructions in case the instruction requires release, acquire or sequential consistency. There is already an `amo_commit_i` signal which indicates that the atomic memory operation finished. If we block the issue stage we need to un-block it if the controller wants to flush.

- **LSU - Data Cache** (`lsu.sv`, `nbdcache.sv`, `cache_ctrl.sv`): Those units will require the biggest amount of changes. Basically every atomic instruction should be treated as a load (because even the stores will return something: success or not). So we need to route those instructions similar to a load. The load unit will interface the dcache with an additional signal `amo_t amo_op_o` indicating whether the current operation is atomic (and which operation) or a regular load.
In case it is an atomic operation the `cache_ctrl ` FSM will enter a new state saving all information (address, op to perform, etc.) and wait for the `amo_commit_i` signal to go high or for `amo_flush_i` to go high. If the latter happens it will return to the idle state. If the `amo_commit_i` signal goes high the cache controller should flush the data cache and issue a new transaction on the bypass signal once the flush is done.
Here we need to differentiate between LR/SC and the AMO OPs. LR/SC will need to use the AXI Lock infrastructure (see AXI Interface specification) to place a reservation (we will need to adapt the testbench for that) while the AMO OPs can retrieve the data process it locally with the AMO ALU. In every case the atomic memory operations will return some result on the `amo_result_o` signal together with a high `amo_valid_o`. Both signals go to the commit stage.
> One further thing which needs a fix is to increase the the width of the dirty SRAM so that it holds a dirty bit per byte. The flush/cache line write-back logic needs to set the corresponding `wstrb` bits. That is because Ariane does not support cache-coherency at the moment and we need to explicitly take care of write-sequentialisation. 

- **Commit Stage** (`commit_stage.sv`): This unit will need to look at the retiring instruction and see that it is an atomic instruction. If it is an atomic instruction the commit stage will assert the `amo_comit` signal for one cycle and block committing new instructions until the `amo_valid_o` is high and indicating a successful atomic memory operation. It will commit the result into the register file. And continue comiting new instructions.

Feel free to check-back, clarify, hack-away.

zarubaf, 2018-03-14:
@raulbehl A coarse description of the problem at hand. I think we can break down the problem in several subtasks of increasing complexity.

raulbehl, 2018-03-14:
@zarubaf, Thank you so much for the detailed description. Please give me some time to get in pace with you as well. :)

raulbehl, 2018-03-16:
Hi @zarubaf, 

I have started going through the A-extensions specifications and have been continuously looking at the Ariane design following your suggestions. For now I am concentrating on the decoder logic/id stage in more detail. Here are my initial thoughts:
-----
As you said most of the AMO related decoding is already in place. I have gone through the code and see that it pretty much handles most of the decoding logic. Though I spotted that the decoder misses to raise an illegal instruction exception when we have non-zero `rs2` field in case of the `AMO_LR` operation. I have added the fix in my local branch for now (still needs testing though).

I had a question regarding the following comment:
> Decoder will also need to block the decoding of any subsequent instructions in case the instruction requires release, acquire or sequential consistency. 

I understand that for AMOs the `amo_commit` signal would only be asserted once the result is written back to the address in `rs1`. I think stalling the decoder till `amo_commit` is asserted might impact performance. As per the ISA the `aq` and `rl` bits places restrictions on memory operations. So, ideally Ariane should be able to process non-hazarding ALU ops and could only stall any new loads/stores entering LSU? Also, I was looking for ways to block decoder (just in case) and could think of the following way: 
- De-assert the `fetch_ack_i` signal until `amo_commit_i` is asserted in the `id_stage`- This would give some back-pressure to the fetch stage but I guess instruction queue should continue to fill-up until full
- De-assert the the `valid` signal for subsequent instructions to avoid processing the same instruction again and again until the atomic is processed

Please let me know if this makes sense. Would be great to get your inputs on this? 

Also, I had the following high level tasks in mind with respect to this:
- Add support to execute AMOs without any of the ordering bits set
- Add support for `aq` and `rl` bits
- Work on adding `LR` and `SC` instructions

Please feel free to update the above tasks accordingly. We could then discuss each of them in more detail (covering both RTL implementation/testbench updates). In the meantime, I would continue investigating. :-)

zarubaf, 2018-03-18:
Hi @raulbehl,

I am absolutely in favour of breaking this up into a couple of smaller manageable steps. It might be a good way that (say once the decoder part is mostly finished) to submit a WIP pull request so that we can keep track of the changes and overall progress as well as do a piece by piece code review there.

> Though I spotted that the decoder misses to raise an illegal instruction exception when we have non-zero rs2 field in case of the AMO_LR operation
Yes I do remember that this was one of my concerns there, very nice.

> I think stalling the decoder till amo_commit is asserted might impact performance. As per the ISA the aq and rl bits places restrictions on memory operations. So, ideally Ariane should be able to process non-hazarding ALU ops and could only stall any new loads/stores entering LSU?

Very good observation! That is absolutely true. At the moment I think it will not have a lot of impact though as we need to flush the cache on every atomic memory operation (maybe not if both `aq` and `rl` are not set). Flushing the cache will take a lot of cycles as we, at least, need to step through every cacheline and look if anything is dirty there. I think the only way this could be more efficiently implemented is through a cache-coherent interconnect which we currently do not have. But yes, we should definitely only block if we encounter another memory op.

> De-assert the fetch_ack_i signal until amo_commit_i is asserted in the id_stage- This would give some back-pressure to the fetch stage but I guess instruction queue should continue to fill-up until full
De-assert the the valid signal for subsequent instructions to avoid processing the same instruction again and again until the atomic is processed

Absolutely correct.

I will start a task list in the initial post. Thanks for your effort, really appreciate it!

Have a nice day,
Florian

P.S.: I think as we are going along and produce some nice documentation by exchanging ideas we should also update the documentation accordingly. Maybe a new `.md` file will do the job where we put our ideas and once we merge the final PR we can clean the documentation up a bit. 

> On that note: We, at ETH, think that cache-coherency in our set-up is not the right way to go. Nevertheless, I am hoping that one day Ariane has at least the option to support cache-coherency. Do you have any recommendation which protocol makes sense to implement? I think that at the end of the day there is only AXI + ACE or TileLink. Any chance that you took a close look at any of those (or both)?

raulbehl, 2018-03-19:
Hi Florian, 

The README part is definitely a good idea. I would start a new `.md` file to capture all this. 

> It might be a good way that (say once the decoder part is mostly finished) to submit a WIP pull request so that we can keep track of the changes and overall progress as well as do a piece by piece code review there.

Yes, this makes sense. It's better to review a bunch of small changes rather than reviewing a big change at once. I'll ensure that I open PRs as in when something takes shape. 

>  Do you have any recommendation which protocol makes sense to implement? I think that at the end of the day there is only AXI + ACE or TileLink. Any chance that you took a close look at any of those (or both)?

I have mostly worked on ARM based systems and hence I am aware of AXI/ACE protocol only. Though given the open nature of the project, I'd suggest taking a look at TileLink too. 

Thank you for clarifying my doubts and adding the task list.

asb, 2018-03-19:
Hi Florian and Rahul. Great to see discussion of adding support for atomics. I've been following along, and am keen to offer whatever support I can. At this stage, just a couple of minor points:
* You should be aware that the memory model working group are tweaking the precise definition of the .aq/.rl/.aqrl annotations on lr/sc/amo*. A draft was [shared in December](https://groups.google.com/a/groups.riscv.org/d/msg/isa-dev/hKywNHBkAXM/wLwNIEpDAwAJ), but there has been further development since then. I believe an updated draft should be shared relatively soon. I don't see this being a blocking issue, but it's something to keep an eye on.
* I'd really recommend adding a sub-task to cover the test and verification story. This can be really subtle, but there are a range of tools designed to help. e.g. [Axe](https://www.cl.cam.ac.uk/~swm11/research/papers/fmcad2016-coherency-checker.pdf) from Cambridge, the [*Check](http://check.cs.princeton.edu/) tools from Princeton, and many others.

An alternate implementation approach could be to start with support for .aqrl, and to treat all variants as if .aqrl was set. This would give a baseline implementation that should be correct (and pass any test you can throw at it), but of course would give stronger semantics than necessary for unordered, .aq, or .rl. You could then add weaker variants.

Although not part of the A extension, you'd probably want to revisit the `fence` instruction as part of this work wouldn't you?

raulbehl, 2018-03-21:
Hi Alex, 

Adding verification related sub-tasks makes sense and should be addressed with this issue itself. 

> An alternate implementation approach could be to start with support for .aqrl, and to treat all variants as if .aqrl was set.

I think this definitely sounds like a better way to approach this. I'll also start looking at the `FENCE` instruction implementation to get some idea of how ordering is currently handled in Ariane. 

Thank you!

zarubaf, 2018-03-21:
Hi,

thanks for the input. I very much agree with the approach to carefully test the implementation with the above mentioned tool. 

I'd also be fine to go with the purely sequential consistency approach in the first place. In any case as long as we don't have a cache-coherent interconnect atomic memory operations will be quite costly. I'd therefore also be very happy to get your input on this (in whatever form: some shared knowledge or actual implementation).

If I have time I will start to check the tools and evaluate them a bit.

> A draft was shared in December, but there has been further development since then. 
The changes should be reflected in the latest ISA Manual, shouldn't they?

Thanks,
Florian

asb, 2018-03-23:
> The changes should be reflected in the latest ISA Manual, shouldn't they?

There are changes after the December update that aren't yet pushed to the ISA manual. The RISC-V Foundation membership agreement sadly makes it difficult to say more, but I understand the group are hoping for another public update soon. I wish that the Foundation had adopted more open development processes, but the memory model group at least seem to be doing a thorough job and making progress.

closed 2018-11-28 by zarubaf