openhwfoundation/cva6 #28: Is elfdpi.h missing from the repo ?

jrrk, 2018-03-21:
elfdpi.h is referenced in elfdpi.cc but doesn't seem to be present or generated anywhere.

raulbehl, 2018-03-21:
Hi Jonathan, 

The file is generated using the following command (makefile should take care of it already):
```
# build TBs
$(library)/.build-tb: $(dpi) $(tbs)
	# Compile top level
	vlog$(questa_version) -sv $(tbs) -work $(library) $(filter %.c %.cc, $(dpi)) -ccflags "-g -std=c++11 " -dpiheader tb/dpi/elfdpi.h
	touch $(library)/.build-tb
```

raulbehl, 2018-03-26:
@zarubaf, @jrrk Do you think it makes sense to add this bit of information somewhere in the README?

jrrk, 2018-03-26:
I believe the correct thing to do would be to group file by simulator. 
For example the majority of RTL files

are simulator independent, but certain files such as .cc testbenches 
will have simulator dependencies. For example

vcs will generate a file called vc_hdrs.h if dpi is used which has a 
similar provenance but possibly works in a different

way. We are unable to match the exact Mentor simulator setup here, but 
ultimately we want users to be able to use

verilator if it is adequate for most tasks. A maintainer equipped with 
all major simulators would be able to

extract the common parts of support files from those which are simulator 
dependent.

zarubaf, 2018-03-26:
Hi,

@jrrk I completely agree. I am also aware that the current solution is not the best one but was a quick way to get started. Not sure whether, fueSoC for example could handle the situation more elegantly.

closed 2018-08-23 by zarubaf