openhwfoundation/cva6 #31: icache corruption

jrrk, 2018-03-28:
When Ariane's instruction and data interfaces are connected to the same memory via a crossbar, you do not want data with a different ID to be returned to the wrong port. However in icache.sv, not only is the master ID 0, but the returned ID is not checked so any old rubbish can get into the icache as below:

--- a/src/icache.sv
+++ b/src/icache.sv
@@ -208,7 +208,8 @@ module icache #(
       end
    endgenerate
 
-   assign refill_ID_to_comp = '0;
+// This master ID is a placeholder. Unique IDs need to come from a central pool ...
+   assign refill_ID_to_comp = '5;
 
    lint_to_axi_refill #(
        .FETCH_ADDR_WIDTH ( FETCH_ADDR_WIDTH ), //= 56,
@@ -310,7 +311,8 @@ module lint_to_axi_refill #(
     assign axi.r_ready   = 1'b1;
 
     assign refill_gnt_o     = axi.ar_ready;
-    assign refill_r_valid_o = axi.r_valid;
+// Don't accept valid signals that don't match our master ID...
+    assign refill_r_valid_o = axi.r_valid && (axi.r_id == refill_ID_i);
     assign refill_r_ID_o    = axi.r_id;
     assign refill_r_data_o  = axi.r_data;
     assign refill_r_last_o  = axi.r_last;
@@ -747,7 +749,7 @@ module icache_controller #(
             if (save_fetch_way)
                 fetch_way_Q <= fetch_way_int;
 
-            //Use this code to be sure thhat there is not apending transaction when enable cache request is asserted
+            //Use this code to be sure that there is no pending transaction when enable cache request is asserted
             if (CS == DISABLED_ICACHE || CS == WAIT_PENDING_TRANS) begin
                 case ({(refill_req_o & refill_gnt_i), refill_r_valid_i})
                     2'b00: begin pending_trans_dis_cache <= pending_trans_dis_cache;       end

jrrk, 2018-03-28:
Sorry that should be:
assign refill_ID_to_comp = 'h5;

zarubaf, 2018-03-28:
I am not sure I get the point here :S

The master is free to choose the ID for the corresponding transaction. The xbar needs to make sure that the answer packages are routed to the right master. The instruction cache just uses one ID for refilling (that is ID 0). Otherwise, if you need to choose a different ID for every master on the xbar, the whole system design is getting quite complex and error prone.

If you want to reduce the three AXI ports of Ariane to just one you need to have an arbiter anyway, which can take care of the ID assignment and ID checking on the return path.

Maybe I am not understanding what you mean (I am actually quite sure that this is the case) so please feel free to correct me.

jrrk, 2018-03-28:
It may be that the crossbar is violating the AXI spec by not filtering the replies. If that is the case its _mea culpa_ but I still say that each master in the Ariane should have a different ID (non zero) to assist debugging in the waveform viewer. I don't know if that is as big an overhead as you say.

zarubaf, 2018-03-28:
Regarding the different IDs I agree as there is no drawback of having it.

closed 2018-08-02 by zarubaf