openhwfoundation/cva6 #33: Should debug_set_pc_i set flush_icache (how else do we inform a new program is loaded ?)

jrrk, 2018-04-06:
// ---------------------------------
        // 1. Exception
        // 2. Return from exception
        // 3. Debug
        // ---------------------------------
        if (ex_valid_i || eret_i || debug_set_pc_i) begin
            // don't flush pcgen as we want to take the exception: Flush PCGen is not a flush signal
            // for the PC Gen stage but instead tells it to take the PC we gave it
            flush_pcgen_o          = 1'b0;
            flush_if_o             = 1'b1;
            flush_unissued_instr_o = 1'b1;
            flush_id_o             = 1'b1;
            flush_ex_o             = 1'b1;
            // flush branch-prediction - it is difficult to say whether this actually loses performance or increases performance
            // because of reduced mis-predicts. There is one case where flushing branch-prediction is absolutely necessary
            // that is when trapping back to machine mode. As the core is making speculative accesses it can happen that it tries
            // to load from an non-idempotent register where a read can have a side-effect. This can happen as the core can try to load
            // from a user-mode address which is then not translated in machine-mode.
            flush_bp_o             = 1'b1;
            **_if (debug_set_pc_i)_**
              **_flush_icache_d         = 1'b1;_**
        end
    end

zarubaf, 2018-04-09:
Yes this is a very good observation and we should definitely add this.

zarubaf, 2018-07-11:
Re-worked debug should fix this #3

closed 2018-08-23 by zarubaf