openhwfoundation/cva6 #41: run dhrystone with questasim

lfernand4, 2018-04-26:
Hi, I am getting the following issue while running dhrystone in questasim 10.6 : 
[image]

[image]

I have generate an elf and an hex.

Can someone help me ?

raulbehl, 2018-04-26:
I guess with the current setup it is not possible to run dhrystone with Questasim as the project currently doesn't support the "printf" function. Basically, the host communication isn't setup yet for Questasim. You should use Verilator in order to run those benchmarks or possible try removing printfs from dhrystone and compiling it using [this](https://github.com/pulp-platform/ariane#running-custom-c-code).

lfernand4, 2018-04-26:
I use verilator : 
[image]
But I have Dhrystones per Second:  1281.0  (0.76D/Mhz)

closed 2018-08-23 by zarubaf