openhwfoundation/cva6 #43: Deadlock on uncached write

jrrk, 2018-05-18:
When writing to memory which outside the cacheable range, this will be served by axi_adapter in miss_handler.sv, but it will be a single write. In the example wrapper, ariane_wrapped.sv, writes are consumed by axi2mem.sv which remains in state WRITE until the last (and only) word is indicated using slave.w_last (actually bypass_if.w_last), likewise axi_adapter remains in state WAIT_B_VALID until axi2bus reaches state SEND_B. However in SINGLE_REQ mode w_last is never set so we have deadlock.

I propose the following patch which prevents the deadlock, I do not know if is functionally correct:

diff --git a/src/miss_handler.sv b/src/miss_handler.sv
index 0c300a9..f494ba6 100644
--- a/src/miss_handler.sv
+++ b/src/miss_handler.sv
@@ -765,7 +765,7 @@ module axi_adapter #(
                             gnt_o = axi.aw_ready & axi.w_ready;
                             gnt_id_o = id_i;
                             case ({axi.aw_ready, axi.w_ready})
-                                2'b11: state_d = WAIT_B_VALID;
+                                2'b11: begin state_d = WAIT_B_VALID; axi.w_last = 1'b1; end
                                 2'b01: state_d = WAIT_AW_READY;
                                 2'b10: state_d = WAIT_LAST_W_READY;
                                 default: state_d = IDLE;

suehtamacv, 2018-05-21:
This might be related to pulp-platform/axi_mem_if#2.

zarubaf, 2018-05-22:
It definitely is. I'll update the submodule. The problem should be fixed.

closed 2018-05-22 by zarubaf