openhwfoundation/cva6 #46: vsim-5 error

mansijain1234, 2018-06-01:
When I am doing run -all command in vsim, I am getting this error.
[image]
Can anyone help me with this?

mansijain1234, 2018-06-20:
Hello,

Actually we were facing this issue when we were trying to run C program with printf and return statements. I think it is because Questa is not linked to fesvr

Thank you,
Mansi

raulbehl, 2018-06-20:
Yes, that is correct. :)

mansijain1234, 2018-06-20:
And one more thing, did you run the CoreMark benchmark on ariane?

raulbehl, 2018-06-20:
No, I haven't got a chance to run much on Ariane lately. I have been busy with some personal work and hoping to get back to speed soon.

mansijain1234, 2018-06-21:
Hello,

Actually I also wanted to know if you found a method to link fesvr with Questasim. If not how are viewing waveforms. Are you dumping the vcd file from verilator and running it on Questasim?

zarubaf, 2018-07-19:
`fesvr` integration is coming with the new debug module #3

closed 2018-08-23 by zarubaf