openhwfoundation/cva6 #47: Error Elaborating Ariane using genus

lvcargnini, 2018-06-05:
Hello, 

I'm trying to synthesize Ariane using Cadence Genus+Innovus+SyntheticLib45nm. So, I have a list of files and during Elaboration phase I'm getting the following error (besides the amount of Wanings of size Mismatch):

zarubaf, 2018-06-05:
Hi,

Yes - I think you are missing the AXI interface. https://github.com/pulp-platform/uvm-components/blob/master/agents/axi_if/axi_if.sv
Regarding the width mismatches they don’t seem critical as they are all in the ariane_pkg but I agree we should get rid of them. Genus should work, I’ve tried a couple of months back.

lvcargnini, 2018-06-05:
Hi,

Thanks,

So should I add this into the 'ariane' or at same level?

Because It was not 'cloned' with other submodules the uvmComponents.

Regarding the widths no, they are not critical, it is just something to look into, if necessary on 'debug' or improvements, but definitely don't affect in anything I agree.

lvcargnini, 2018-06-06:
the UVM components were the missing component.

closed 2018-06-06 by lvcargnini