openhwfoundation/cva6 #48: Forwarding of FP registers

suehtamacv, 2018-06-27:
There is a potential problem with the forwarding logic of floating point registers.

https://github.com/pulp-platform/ariane/blob/8620f41ea1db93c7883844011092147a23cabab3/src/scoreboard.sv#L217-L220

In the excerpt above, it is the destination register of `mem_q[i]` that is being forwarded as source register of `issue_instr_o`. Because of this, I think the correct check should be `is_rd_fpr(mem_q[i].sbe.op) == is_rs1_fpr(issue_instr_o.op)` instead. There are more cases of this throughout the scoreboard.

---

Also, in `is_rd_fpr` (and `is_rs2_fpr`), shouldn't the test below go up to `[FLD:FLB]` instead of stopping at `FLW`?

https://github.com/pulp-platform/ariane/blob/8620f41ea1db93c7883844011092147a23cabab3/include/ariane_pkg.sv#L253-L256

zarubaf, 2018-06-27:
The same for then, right?
https://github.com/pulp-platform/ariane/blob/8620f41ea1db93c7883844011092147a23cabab3/include/ariane_pkg.sv#L230

zarubaf, 2018-07-03:
@suehtamacv  The issue should be solved. Would appreciate short feedback. Lets leave this open for later reference (e.g.: constructing a test case).

PR #51 (2018-07-03): Fix FP forwarding and missing FP loads/stores
Fixes #48
closed 2018-09-23 by zarubaf