openhwfoundation/cva6 #50: Issue while compile and link of ceil,scanf ,free or malloc function in C

ShravanSubbaraman, 2018-07-03:
[image]

Hi i am facing this issue when trying to run functions that are not defined in the syscalls.c file. 

But the issue is actually telling me that this is a memory displacement issue. Is there a way to solve this problem? 

So is it like the riscv-tools/riscv-tests/benchmarks/common/syscalls.c contains only certain limited functions like printf and string related functions. If in case i need to run some data structures then  should i modify the syscalls according to my need?

zarubaf, 2018-07-03:
Have you used the same compiler for compiling `newlib` as you are using here (might be something related to compressed instructions)?

I am actually not sure what the best way of handling the whole syscall business is. But probably implementing as much as possible not relying on `fesvr`.

zarubaf, 2018-07-03:
Maybe Germain can help here? @haugoug

closed 2018-08-23 by zarubaf