openhwfoundation/cva6 #60: Error in "make sim"

redtomato11, 2018-07-16:
Hi,

I am getting following error in "make sim" in the transcribt window of questasim:

# do tb/wave/wave_core.do
# ** Error: (vish-4014) No objects found matching '/core_tb/dut/i_frontend/*'.
# Error in macro ./tb/wave/wave_core.do line 3
# (vish-4014) No objects found matching '/core_tb/dut/i_frontend/*'.
#     while executing
# "add wave -noupdate -group frontend /core_tb/dut/i_frontend/*"

Can you please help me with that?
Tobi

zarubaf, 2018-07-18:
I see, that seems to come from inconsistencies from the Git submodules. I should resolve this in the next weeks when I am merging the latest changes.

For now I see two options for you:

1. You switch to the stable dev branch called `ariane_next`
2. You delete the offending lines in  `wave_core.do`. It does not impact functionality but just won't log the wave-forms.

zarubaf, 2018-07-24:
Should be resolve now.

closed 2018-07-24 by zarubaf