openhwfoundation/cva6 #62: HPTW

ShravanSubbaraman, 2018-07-18:
Why does both the TLBs get updated from the data read from the DCache?

[image]

Here the pte is nothing but the data read from the DCache. Then when the PTW goes for a walk to perform the PTE lookup then aren't we looking at the wrong entry for the ITLB ?
Can you explain it please?

ShravanSubbaraman, 2018-07-20:
I get the point with regard to common PTE. But when you have a instruction miss in your TLB why do you want to go through you dcache and find it ? So now your dcache contains both data and instruction information as well ? Then why do we need a separate I$ then ? Because if we are performing address translation even for the IF stage then possibly there will be misses right so my question is why are you handling it through dcache ? The final address is found by accessing through the dcache and not the icache. 

Correct me if am wrong.

zarubaf, 2018-07-21:
Because there is no way in RISC-V to sync address translation data (PTEs) and instruction stream (icache). Also why would you want to go through the instruction cache? It makes no sense to fetch one PTE from the icache and the other from the dcache. I do not get your point.

ShravanSubbaraman, 2018-07-21:
Thanks Florian ! I think i was able to connect the dots. I was just wanting to know why you wanted to check the pte only through the D$ and not the I$.

closed 2018-07-24 by zarubaf