openhwfoundation/cva6 #63: Address translation in MMU

ShravanSubbaraman, 2018-07-18:
I want to perform address translation using the MMU. Currently my core is running in M mode wherein i don't have the capability to perform the address translation. 

What all changes has to be done to get the address translation to work. Like where in the ISA should i change? Are there any changes to the core that i have to do with respect to this?

zarubaf, 2018-07-18:
First, why would you want to do address translation in machine mode, claryfing your use-case may help me to answer you.
RISC-V does not allow you to translate instruction fetch requests in M mode. You can do translated load/stores though if you set the `mprv` field in `mstatus`. The privilege level with which address translation is done is determined by the value in `mpp`. 

https://github.com/pulp-platform/ariane/blob/a006526c6f1b9cab24dc6b39a672036c7f014f46/src/csr_regfile.sv#L463

I suggest you take a closer look at the RISC-V privileged specification.

ShravanSubbaraman, 2018-07-18:
I am not trying to perform the address translation in the M mode. I want to visualize the address translation feature to be working in my MMU. For which i will have to switch modes. Now to do that what are the changes that i have to incorporate?

zarubaf, 2018-07-19:
That is nothing specific to Ariane and you can figure this out by reading the RISC-V privileged spec. But basically you need to set `satp`, set `mpp` to U-mode (which it should be by default after reset) and do an `mret`. ISA questions like those are better suited for the RISC-V mailing lists.

closed 2018-07-19 by zarubaf