openhwfoundation/cva6 #77: FPGA Synthesis Timing issues

lvcargnini, 2018-08-20:
Hello, 

Not sure, since there are no forums: I'm trying to synthesize it for an FPGA Xilinx xcvu9p* family,  I created a design block that connects the three AXI, to access brams, the goal it is evaluate if I can synthesize it for >= 200Mhz of frequency. 

However I'm facing a barrage of timing issues, that weren't expected, for things like, input and output pads:
...
        input  logic                           test_en_i,     // enable all clock gates for testing
        // CPU Control Signals
        input  logic                           fetch_enable_i,
        // Core ID, Cluster ID and boot address are considered more or less static
        //input  logic [63:0]                    boot_addr_i,
        input  logic [ 3:0]                    core_id_i,
        input  logic [ 5:0]                    cluster_id_i,
        input  logic                           flush_req_i,
        output logic                           flushing_o,
        // Interrupt inputs
        input  logic [1:0]                     irq_i,        // level sensitive IR lines, mip & sip
        input  logic                           ipi_i,        // inter-processor interrupts
        output logic                           sec_lvl_o,    // current privilege level oot
        // Timer facilities
        input  logic [63:0]                    time_i,        // global time (most probably coming from an RTC)
        input  logic                           time_irq_i,    // timer interrupt in
        // Debug Interface
        input  logic                           debug_req_i,
        output logic                           debug_gnt_o,
        output logic                           debug_rvalid_o,
        input  logic [15:0]                    debug_addr_i,
        input  logic                           debug_we_i,
        input  logic [63:0]                    debug_wdata_i,
        output logic [63:0]                    debug_rdata_o,
        output logic                           debug_halted_o,
        input  logic                           debug_halt_i,
        input  logic                           debug_resume_i,
 ...

The boot address I fixed:
                .boot_addr_i(64'h8000_0000),  
The goal here is synthesize it on an FPGA, to prove that this same core can be used on FPGA, ... I understand that it looks reverse path, but yes, on ASIC I'm sure on what to do, and I got the timing closure on ASIC flow. However in FPGA on Vivado it is being quite cumbersome, what have I forgot to do or add?

By the way I had to made a slight change in the code to use synchronous reset, otherwise, I was getting nowhere with timing in the FPGA. So I created a serieis of ifdefs to choose accoridng with a  MACRO modifier what to use:
ifdef PULP_FPGA_EMUL
always @(posedge clk) 
else
always @(posedge clk or negedge rstn)
endif
I'm using the PULP_FGPA_EMUL, since I had to enable it to synthesize it for FPGAs.

I have an oscillator of 300Mhz, that gets into a PLL,  and generates a 200 and 250 Mhz signals, that are the ones I'm using to feed the system.

SDC/XDC clock constraints:
create_clock -period 3.333 [get_ports clk_in1_p]
set_input_jitter [get_clocks -of_objects [get_ports clk_in1_p]] 0.033330000000000005

set_property PHASESHIFT_MODE LATENCY [get_cells -hierarchical *adv*]

//set the reset in a false path to prevent the timing report to goes bananas on the reset signals
set_false_path -to [get_pins -hier *cdc_to*/D].

There is a forum to discuss design issues like this one?

Regards,
Vitorio.

zarubaf, 2018-08-20:
Hi Vitorio,

thanks for the feedback. I think here is actually the right place to discuss such things. We do not have a separate forum. If you want you can also send such issues to our mailing list (see https://www.pulp-platform.org/faq/ on how to subscribe).

1. The asynchronous reset (active low) is something we've been using in most of our PULP project hence we also use it in Ariane. I am not awfully familiar with the reasons and I definitely see that it makes the FPGA flow more cumbersome. So from my point of view that is definitely something we can discuss.  Especially as we are currently trying to map a system of a couple of Ariane cores to FPGA. @msfschaffner @gurkaynak may provide more insight.

2. Regarding the input/output signals. Where do you map them - directly to/from pads? I would suggest you connect them to registers or drive them statically. That should not give any timing problems.

Regarding the interrupts: The core does not include synchronizers so you need to have them externally.

Are you able to close timing at that frequency on your device if you disregard the IO paths? I do remember that I was closing timing at approximately 50 MHz on a Virtex 7. Don't know how that compares to your device though.

That being said, we are likely to release a very small SoC which provides all the infrastructure together with an FPGA flow in the next month(s).

lvcargnini, 2018-08-20:
Hi Florian,

Thanks for the feedback. I'll look into the mailing list. That way I can look for help there.

- The asynchronous low phase resets are not a problem. The problem it is, and it mentioned to me before for a former colleague back in grad school, the asynchronous signals on FPGA are an invitation for a messy situation, hence I discovered that with Ariane. There is maybe a 'magic enchantment' on the SDC/XDC to ignore this or force it to map on asynchronous FLIP-FLOP the FDPE on Xilinx, however making a dirty change on the code in regards the sensitivity list due to a macro, was simpler and it works.
- The signals: yes directly in pads, I'll try to pass through a register first, haven't taught about that, thanks.

Interrupters, no worries, I'm trying to synthesize it for an FPGA first thing, the next steps it is integrating it with a larger design block, step-by-step progression here.

I'll sign the list and we get in touch there.

Regards,
Vitorio.

lvcargnini, 2018-08-20:
Re-checking my top, I set the input/output this way:

```
module top_ariane_simple #(
parameter logic [63:0] CACHE_START_ADDR  = 64'h8000_0000, // address on which to decide whether the request is cache-able or not
        parameter int unsigned AXI_ID_WIDTH      = 10,
        parameter int unsigned AXI_USER_WIDTH    = 1,
        parameter int unsigned AXI_ADDRESS_WIDTH = 64,
        parameter int unsigned AXI_DATA_WIDTH    = 64
       // parameter TRANS_ID_BITS = ariane_pkg.TRANS_ID_BITS
)
(

        input  logic                           test_en_i,     // enable all clock gates for testing
        // CPU Control Signals
        input  logic                           fetch_enable_i,
        // Core ID, Cluster ID and boot address are considered more or less static
        //input  logic [63:0]                    boot_addr_i,
        input  logic [ 3:0]                    core_id_i,
        input  logic [ 5:0]                    cluster_id_i,
        input  logic                           flush_req_i,
        output logic                           flushing_o,
        // Interrupt inputs
        input  logic [1:0]                     irq_i,        // level sensitive IR lines, mip & sip
        input  logic                           ipi_i,        // inter-processor interrupts
        output logic                           sec_lvl_o,    // current privilege level oot
        // Timer facilities
        input  logic [63:0]                    time_i,        // global time (most probably coming from an RTC)
        input  logic                           time_irq_i,    // timer interrupt in
        // Debug Interface
        input  logic                           debug_req_i,
        output logic                           debug_gnt_o,
        output logic                           debug_rvalid_o,
        input  logic [15:0]                    debug_addr_i,
        input  logic                           debug_we_i,
        input  logic [63:0]                    debug_wdata_i,
        output logic [63:0]                    debug_rdata_o,
        output logic                           debug_halted_o,
        input  logic                           debug_halt_i,
        input  logic                           debug_resume_i,
 
  input refclk_300mhz_clk_n,
  input refclk_300mhz_clk_p,
  input logic sysrstn

);
```

So the 'logic' keyword should enable the synthesizer to automatically add the register, I can see on the schematic that it is not the case, so I'll force and manually add the registers.

Vitorio.

zarubaf, 2018-08-20:
I don't think that `logic` adds registers, its just a 4-state wire.

lvcargnini, 2018-08-20:
Before SystemVerilog, in Verilog if you remember, you add a input or output to register it, you need to repeat the signal with the `reg` keyword, according to with SystemVerilog spec, from 2011, I think, the goal was to use the keyword `logic` on the module port declaration to allow the synthesizer to infer registered input and/or outputs. However it doesn't seem to be the case, ... anymore.

Anyway, I added registers, manually, Plus the `logic` should replace wire and reg keywords and 'magically' solve all the incongruences from the old wire vs reg.

zarubaf, 2018-08-29:
The architecture didn't change. Main changes are code clean-up, riscv compliant debug and re-structuring the cache hierarchy. A lot of things have changed at the TB though and the interface has been simplified (removing the need for external cache flushing because of the new debug module). I'd personally suggest to go for the update. As far as I understood it you have changed:

- Adding registers to the interface (you just need to remove signals there to be compliant)
- Replaced reset logic with sync active high (you can do that with a search and replace probably)

If you did other architectural changes then the story will get a little bit more complicated I guess :( 

Of course you are also more then welcome to keep what you've done but it will get more and more difficult to merge patches and improvements (which for sure will come in the future). That is always a thing with out of tree forks unfortunately. At a certain time in the future we can certainly talk about the reset story, I am not entirely convinced (and I am honestly missing the expertise here) that active low asynchronous is really clever.

P.S.: The changelog (although not perfectly accurate in detail gives some more insights regarding the changes: https://github.com/pulp-platform/ariane/blob/master/CHANGELOG.md)

msfschaffner, 2018-08-30:
Hi @lvcargnini,

Since we use FPGAs mainly as an emulation platform at the moment, we keep the asynchronous reset strategy for convenience. It is true that there can be an impact on timing and resource utilization on certain devices, but this is usually quite small (~4% slower, <0.5% more logic in case of the current Ariane design). 

In terms of operating frequency, I expect the Ariane design to close around 150MHz on a Virtex UltraScale+ device (SG2).

We might add some FPGA specific optimizations in the future, and a change of the reset strategy could be part of this (but I can not promise anything at the moment).

gurkaynak, 2018-08-30:
Actually let me weigh in a little bit for the active-low asynchronous reset for a bit. There are plenty of things that certain FPGA vendors implement on their devices, and if designs are being made in the way that these vendors have chosen to do, these designs can be mapped more easily. This is not only limited to reset, and as FPGA vendors start adding more and more resources such 'choices' will just increase. It is fine if your design is explicitly made to be mapped to this specific vendor. But if you think about that there are many different vendors, and many different 'valid' implementation choices, there is no 'reason' to favor one choice over the other one. 

For ASIC design, resets have to be asynchronous. I can discuss the issues at length if you want. Whether or not these are active high or low, can be debated, but I have not yet heard a technical argument that could demonstrate that high active reset is better than low active. If anything, a small standard cell library is more likely to have an active low reset FF than an active high one (due to same historic reasons that we also use). At the same time, ever since we started using CMOS logic, there is no real reason that the polarity of rarely activated signals makes it better or worse. 

As long as the code base is consistent, it should be possible to replace the reset to whatever you need fairly easily. However, I can clearly state that the reset signal will not change in our designs in the near future.

zarubaf, 2018-08-30:
> For ASIC design, resets have to be asynchronous. I can discuss the issues at length if you want.
Yes please :raising_hand_man:

gurkaynak, 2018-08-30:
Meet me in my office next week :)

lvcargnini, 2018-08-30:
Hi @msfschaffner,

Don't worry about the FPGA design friendliness. I made a PR today adding a DFF and register module for Ariane, this can be used to 'automagically' translate into FDCE on Xilinx Async DFF, or to a DFF on IC flow, I know it worked on experiments. I also added a register module based on it. 
I also added two generic SRAM modules, that on FPGA translate into BRAMs 'automagically' too. However needs some work for IC-flow yet (Not there).

The biggest issue on FPGA: it has its quirks, I was working on IC when someone asked to see it working on FPGA, that it is how I ended up doing this. So don't worry, keep the focus on RTL design, independent of a target platform. It is doing the great the project so far.

@gurkaynak Thanks, not discussing the ASIC / IC flows/target. I know that. Don't worry, please don't change the design. It is necessary on FPGA SDC/XDC to add a few 'magic' constraints for the tool flow ignore or not complain of asynchronous signals. Or in my specific case, I had to force it to understand to use FDCE (Xilinx async DFF), however, the same trick works on IC flow for general DFF of the foundry DK/iPDK.

lvcargnini, 2018-08-30:
It just needs to be worked around the FPGA quirks, no need to change the design, there are no issues, it is all FPGA vendors tool flow.

closed 2018-08-30 by lvcargnini