openhwfoundation/cva6 #78: Failed on verilator

kuopinghsu, 2018-08-21:
When I run the riscv-tests/benchmark on the verilator, I got following message

*** FAILED *** (tohost = 1337)

This seems to have received an illegal instruction. This is part of the log of the trace file.

```
               20691 0x800010ea (0x05500613) DASM(05500613)
               20692 0x800010ee (0x00900513) DASM(00900513)
               20693 0x800010f0 (0xfdd6879b) DASM(fdd6879b)
               20695 0x800010f4 (0x0ff7f793) DASM(0ff7f793)
               20696 0x800010f8 (0x00170413) DASM(00170413)
               20697 0x800010fc (0x1ef66963) DASM(1ef66963)
               20698 0x80001100 (0x00279793) DASM(00279793)
               20700 0x80001102 (0x015787b3) DASM(015787b3)
               20704 0x80001104 (0x0007a783) DASM(0007a783)
               20706 0x80001106 (0x015787b3) DASM(015787b3)
               20707 0x80001108 (0x00078067) DASM(00078067)
Exception Cause: Illegal Instructions, DASM(0000200b)
               20731 0x80000120 (0xef010113) DASM(ef010113)
               20734 0x80000122 (0x00113423) DASM(00113423)
               20736 0x80000124 (0x00213823) DASM(00213823)
               20737 0x80000126 (0x00313c23) DASM(00313c23)
```

This is part of the assembly code,

```
    800010f0:	fdd6879b          	addiw	a5,a3,-35
    800010f4:	0ff7f793          	andi	a5,a5,255
    800010f8:	00170413          	addi	s0,a4,1
    800010fc:	1ef66963          	bltu	a2,a5,800012ee <vprintfmt+0x256>
    80001100:	078a                	slli	a5,a5,0x2
    80001102:	97d6                	add	a5,a5,s5
    80001104:	439c                	lw	a5,0(a5)
    80001106:	97d6                	add	a5,a5,s5
    80001108:	8782                	jr	a5
    8000110a:	60f6                	ld	ra,344(sp)
    8000110c:	6456                	ld	s0,336(sp)
```

Everything is fine when I get the code back to hash a193182.

zarubaf, 2018-08-21:
Thanks, we were able to reproduce and have localized an internal commit which is likely to have caused this bug. I will keep you updated. We are hopefully able to fix it in the next 24 hours. Thanks again for pointing out so quickly.

zarubaf, 2018-08-22:
Okay, that should be fixed now. Note: some benchmarks in RISC-V benchmarks will always fail as they use floating point operations which we currently do not support in our open-source version.

kuopinghsu, 2018-08-23:
Thanks for your great work. There's an another illegal instruction exception on the attached file.

[benchmarks.zip](https://github.com/pulp-platform/ariane/files/2312870/benchmarks.zip)

zarubaf, 2018-08-23:
Hm, okay. So my suggestion would be to go with the [ariane_next](https://github.com/pulp-platform/ariane/tree/ariane_next) branch for the moment. I am actually a bit too lazy to try to reproduce the problem on code which is anyway replaced in the next hours 😅On the branch we do have the benchmarks as CI tests and its much more cleaned-up. I put the the branch up for review and we might be able to merge it into master in the next hours.

commit db77f10b (2018-08-22): Fix #78 (#80)

Fix #78
commit 5e773455 (2018-08-22): fix #78
commit 635de4bc (2018-08-22): Merge pull request #82 from msfschaffner/ariane_next

- Fix typos in clint
- Re-structure CI flow
- Fix #78
closed 2018-08-23 by msfschaffner