openhwfoundation/cva6 #81: using vivado to package ariane as an ip core

rgaehfgoiarewhg, 2018-08-22:
i am trying to create an ip core with vivado of the ariane cpu but i am getting strange errors regarding - among others - the following code segment:

```
AXI_BUS
     #(
       .AXI_ADDR_WIDTH (AXI_ADDRESS_W),
       .AXI_DATA_WIDTH (AXI_DATA_W   ),
       .AXI_ID_WIDTH   (AXI_ID_IN    ),
       .AXI_USER_WIDTH (AXI_USER_W   )
     )
     axi_slave[N_SLAVE_PORT]();
```

apparently vivado does not like this very much. it looks like it is missing a flie in the hierarchy treeview. 
[image]

also, is ariane actually usable? like ready to be used for something? the documentation looks like it is only some prototype.

thank you very much in advance!

zarubaf, 2018-08-22:
You are missing the SystemVerilog interfaces of the AXI Bus. https://github.com/pulp-platform/uvm-components/blob/6bb81367a35bde8952217685dba9457183bc63ea/agents/axi_if/axi_if.sv
It might be possible that Vivado doesn't like interfaces on a top-level. You have to pack those in a wrapper then.

What do you mean by actually useable? Its a 64-bit RISC-V core with RV64IMC + full privileged spec, not more not less. What parts in the documentation gives you the impression? I'd appreciate constructive feedback.

rgaehfgoiarewhg, 2018-08-22:
thank you very much for the quick reply. i will try it.

i was unsure how far along the project actually is.
there is very little documentation and a lot is focused on running simulations. this gave me the impression that it might not be ready yet.

zarubaf, 2018-08-22:
You have seen https://pulp-platform.github.io/ariane/ ? Let me know where the documentation doesn’t answer your questions.

rgaehfgoiarewhg, 2018-08-23:
where is it stated that i need that file you mentioned?

zarubaf, 2018-08-23:
My suggestion is you look at whats being compiled for verilator and use the same files.

rgaehfgoiarewhg, 2018-08-23:
as i said: great documentation!