openhwfoundation/cva6 #109: data_if & bypass_if

suppamax, 2018-10-05:
Hi,

I am currently using your Ariane version a19318201a368c1e31c384366549998c964744df.
I see that the cache controller in nbdcache.sv has the following connection
                .bypass_i              ( ~enable_i            ),
which results, if I understand correctly, in disabling the data_if port and consequently the D$.
How to make a proper use of this interface?
And how should I drive the flush_dcache_i signal at top level?

Thanks,
Max

zarubaf, 2018-10-06:
Yes in principle that would be the intended behaviour, but unfortunately doesn't work #68 
I've never been in need for disabling the data cache. `flush_dcache_i` can be driven low if you do not plan to flush the caches from external. This interface has been removed in the newer versions as we are supporting the RISC-V debug specification.

suppamax, 2018-10-07:
Hi,
thanks for the reply but still this is not clear to me.
In simulation I see that data_if is never used, because of the ".bypass_i ( ~enable_i)" setting mentioned before, all data are accessed via bypass_if.
But let's assume the D$ is enabled: how is the write policy?

I'll move to the latest version, to check the differences.

zarubaf, 2018-10-08:
There is a cache-base address (decision address) parameter when you instantiate Ariane. If the load physical address is above this threshold the access is cached, if not the address is access via bypass if. The D$ is enabled by default and disabling currently doesn't work, sorry. So you basically need to design your address map so that the periphery is below this threshold. Don't hesitate to ask if this doesn't clear things up.

zarubaf, 2018-10-16:
Closing due to inactivity. I assume the answer was what you were looking for. If not feel free to re-open.

closed 2018-10-16 by zarubaf