openhwfoundation/cva6 #117: [Enhancement] Micro TLB + Multilevel TLB

msfschaffner, 2018-10-16:
Splitting the TLB into a micro TLB with 4 entries and a larger (possibly multilevel) TLB can improve the timing on the cache tag comparison paths.

MikeOpenHWGroup, 2023-02-18:
@zarubaf, was is the status of this enhancement? I recommend we either:
1. Set the `Status:Wont-fix` label and close it, or
2. Assign this issue to @JeanRochCoulon and ask for him to find an owner for this enhancement.

jquevremont, 2023-02-21:
@sjthales is working on this topic.

zarubaf, 2023-10-20:
This support has been merged for the 32-bit version. Not planned for 64-bit at the moment, afaik. If it is than we should add it as a new task.

closed 2023-10-20 by zarubaf

jquevremont, 2023-10-23:
> This support has been merged for the 32-bit version. Not planned for 64-bit at the moment, afaik. If it is then we should add it as a new task.

To anyone who reads this: If you intend to do this, come and synchronize with the CVA6 team.
The context is the following: we have several MMUs in the repo and we'd like to refactor them.