openhwfoundation/cva6 #134: trace log needs improving

jrrk, 2018-11-15:
ariane produces a tracelog trace_core_00_0.dasm, which may be converted to trace_core_00_0.dis using spike-dasm. I have found that the trace log disagrees with the disassembled binary in the sense that compressed instructions are shown post conversion:

disassembly:
0000000040000382 <_init>:
    40000382:   00001517                auipc   a0,0x1
    40000386:   2c650513                addi    a0,a0,710 # 40001648 <_bss>
    4000038a:   00001617                auipc   a2,0x1
    4000038e:   30e60613                addi    a2,a2,782 # 40001698 <_end>
    40000392:   1101                    addi    sp,sp,-32
    40000394:   8e09                    sub     a2,a2,a0

trace log:
                  88 0x40000382 M (0x00001517) auipc   a0, 0x1
                  89 0x40000386 M (0x2c650513) addi    a0, a0, 710
                  90 0x4000038a M (0x00001617) auipc   a2, 0x1
                  91 0x4000038e M (0x30e60613) addi    a2, a2, 782
                  92 0x40000392 M (0xfe010113) addi    sp, sp, -32
                  93 0x40000394 M (0x40a60633) sub     a2, a2, a0

also sometimes the expanded instruction is < 0xFFFF leading to an ambiguity
for poor humans to get confused by:

    40000068:   6319                    lui     t1,0x6
 
becomes:

                  65 0x40000068 M (0x00006337) lui     t1, 0x6

In addition the PC counter log appears misaligned to those not in the know
because 32-bit instructions are apparently fetched with a PC increment of 2.

zarubaf, 2018-11-16:
I agree regarding the compressed instructions. I think we need to think about generating the tracer defines out of the riscv opcodes repo. Thanks for pointing this out.

zarubaf, 2018-12-08:
So the verilator log actually uses `spike-dasm` which understands the aliases (its also generated from the opcodes directory). The Questa tracer still uses the expanded instructions. I actually wanted to generate the tracer automatically from the opcodes directory, then I saw that you already forked it and abandoned the mission because I had a different understanding on which verilog to generate.

So what would be needed it to generate the trace defines automatically and add our non-standard floating point instructions. Also the aliases would be nice to have in QuestaSimulation (for the moment I hardcoded some of them).

closed 2019-06-04 by zarubaf

Yi1101, 2020-07-23:
@jrrk 
Dear Jrrk,
Where did you find the trace_core_00_0.dasm file. When ran Questasim simulation, which only generated trace_hart_0_commit.log and trace_hart_0.log files. Then I extracted hex_value from file trace_hart_0_commit.log, and used Spike-dasm to translate it into disassemble code. Finally, compare with original disassemble code from ELF file. Even though, most of the code are the same, but some of them are totally not the same. Do you have any ideas?

jrrk, 2020-07-23:
I assume you are referring to Ariane. This cpu decompresses 16-bit instructions to 32-bit before it reaches the trace log. There is not enough info to reconstruct the original elf file code at the moment. If you compile without compressed instructions I predict you will not get a mismatch.

commit 9c800a67 (2019-04-20): instruction_tracer: Fix compressed tracing (fix #134)

The built-in tracer now supports tracing compressed instructions. The
Spike commit log feature has been adapted and is now a 100% aligned.
commit 518f44d4 (2019-04-24): instruction_tracer: Fix compressed tracing (fix #134)

The built-in tracer now supports tracing compressed instructions. The
Spike commit log feature has been adapted and is now a 100% aligned.
commit 1f2aeeb1 (2019-04-24): instruction_tracer: Fix compressed tracing (fix #134)

The built-in tracer now supports tracing compressed instructions. The
Spike commit log feature has been adapted and is now a 100% aligned.
commit 4c995a41 (2019-04-25): instruction_tracer: Fix compressed tracing (fix #134)

The built-in tracer now supports tracing compressed instructions. The
Spike commit log feature has been adapted and is now a 100% aligned.