openhwfoundation/cva6 #136: Highest bits in Sv39

janhoogerbrugge, 2018-11-20:
The spec says:

  Load and store effective addresses, which are 64 bits, must have bits 63–39 
  all equal to bit 38, or else a page-fault exception will occur.

The code says:

  // check that all bits in the address >= 39 are equal
  if (!((&lsu_ctrl.vaddr[63:39]) == 1'b1 || (|lsu_ctrl.vaddr[63:39]) == 1'b0)) ...

So then the 39 in the code should be 38?

Jan

zarubaf, 2018-11-20:
Yes you are absolutely right. Unfortunately there is some other issue with this code: Actually we should not test this in machine mode. @msfschaffner has a patch for this and will also include the issue you address. Thanks for bringing this up.

msfschaffner, 2018-11-20:
Thanks for reporting this. I'll include it in the patch that fixes the page-fault checks.

PR #138 (2018-11-22): Fix mismatches in virtual address checking logic (this fixes #136).
- Instr/ld/st must only throw access faults when virtual memory translation is enabled
- Correct tested bit slice from [63:39] to [63:38]
- Fixes #136
commit 190416b7 (2018-11-22): Merge pull request #138 from msfschaffner/ariane_next

Fix mismatches in virtual address checking logic (this fixes #136).

- Instr/ld/st must only throw access faults when virtual memory translation is enabled
- Correct tested bit slice from [63:39] to [63:38]
- Fixes #136
closed 2018-11-22 by msfschaffner