openhwfoundation/cva6 #137: make verilate error

merchantf, 2018-11-21:
Hi,

I am getting following error. 

%Error: src/fpga-support/rtl/SyncSpRamBeNx64.sv:136: syntax error, unexpected assert

Any idea why am I getting this error?

Farhad

msfschaffner, 2018-11-22:
Hello Farhad,

It would be helpful if you could specify which branch you are on, and which Verilator version you are using.

Please make sure you check out the newest version of Ariane and its submodules (`git submodule update --init --recursive`) and try again. Also make sure that you use the same Verilator version that we use (3.924).

Michael

merchantf, 2018-11-26:
> Hi Farhad,
>
> Does this issue still persist? Otherwise I would like to close this.
>
> Michael

Hi,

For now no. If I get any other problem then I will post it. Thanks a lot.

I am facing some problem in pulpino. Who should I contact?

Farhad

msfschaffner, 2018-11-26:
Just open up an issue on the pulpino repo page.

Thanks, 
Michael

closed 2018-11-26 by msfschaffner