openhwfoundation/cva6 #147: 1801c5e0ea231f83e9ba5422b9a7a4feaaad879f is not a tree

jrrk, 2018-11-27:
ariane_next branch points to revision 1801c5e0ea231f83e9ba5422b9a7a4feaaad879f

ariane_next$ git submodule update --recursive
fatal: reference is not a tree: 1801c5e0ea231f83e9ba5422b9a7a4feaaad879f
Unable to checkout '1801c5e0ea231f83e9ba5422b9a7a4feaaad879f' in submodule path 'src/fpu'

but this does not exist. It should point to heads/master. Perhaps somebody did git add in the top-level and pushed without pushing submodule src/fpu to the remote ?

zarubaf, 2018-11-27:
Hm, actually https://github.com/pulp-platform/fpnew/tree/1801c5e0ea231f83e9ba5422b9a7a4feaaad879f exists :/ Maybe try deleting the

jrrk, 2018-11-27:
OK, so there is a disagreement between GUI and command line. I have 
forked my own copy and patched your changes into master here:

https://github.com/pulp-platform/fpnew/compare/master...jrrk:master

I don't know what you will do with it though if you can't reproduce my 
issue.

asb, 2018-11-28:
I believe the issue is that the commit isn't present on any branch currently on github (maybe it was pushed and later deleted?):

```
git clone https://github.com/pulp-platform/fpnew
Cloning into 'fpnew'...
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 603 (delta 0), reused 2 (delta 0), pack-reused 600
Receiving objects: 100% (603/603), 262.05 KiB | 820.00 KiB/s, done.
Resolving deltas: 100% (434/434), done.
asb@pudding /tmp $ cd fpnew/
asb@pudding /tmp/fpnew $ git show 1801c5e0ea231f83e9ba5422b9a7a4feaaad879f
fatal: bad object 1801c5e0ea231f83e9ba5422b9a7a4feaaad879f
```

Suggested resolution: push a branch to GitHub which contains this commit.

zarubaf, 2018-11-28:
Indeed, very strange... Working on it.

msfschaffner, 2018-11-28:
That commit seems to point to a commit that we have locally on our
GitLab server. We'll fix that in a minute.

Florian & Michael

zarubaf, 2018-11-28:
This is now fixed on the latest ariane_next. Thanks a lot for reporting this and helping with streamlining. Sorry for the rough experience.

closed 2018-11-28 by zarubaf

jrrk, 2018-11-28:
Thanks, I do not see the error any more.