openhwfoundation/cva6 #153: [Enhancement] Add RVFI

zarubaf, 2018-12-04:
Add support for RISC-V formal interface.

raulbehl, 2019-05-07:
Hi Florian, 

I know you guys have a lot on-going but any heads up on when this would be done? 

Thank you, 
Rahul

zarubaf, 2019-05-07:
Hi Rahul,

unfortunately, at this stage, we don't have any further information on when we can tackle to add this feature. There are a couple of other fixes planned first, so this probably has to wait for a little longer (at least from our side).

Best,
Florian

raulbehl, 2019-05-08:
Sure. I understand. I have actually used the basic RVFI in one of the project and can help with the integration. Though I would need help to capture the right RTL state as needed by the interface. 

Do you think that can be done?

hyunsuchae, 2020-02-05:
Hi, I am trying to add RVFI feature to Ariane, if there isn't a version of this yet.
Is there a way for me to get the SystemVerilog code that does not have cache integrated in Ariane?
Thanks.

MikeOpenHWGroup, 2023-02-18:
RVFI has been supported for some time.  Closing.

closed 2023-02-18 by MikeOpenHWGroup