openhwfoundation/cva6 #157: The CPU is always in debug mode during emulation

TianyangL, 2018-12-25:
I use “work-ver/Variane_testharness rv64um-v-divuw” ,But the CPU is always in debug mode during emulation.
Why?

msfschaffner, 2019-01-07:
Hi TianyangL, 

The CPU is mostly in debug mode, yes. This is due to the mechanism that is used to load the executable (we make use of the RISCV FESVR, which uses the debug module to access system memory). 

On close examination of your `trace_hart_00.dasm` you should see the test starting towards the end of the trace, around timestamp 775668 as shown below:

```
...
775645 0x854 D (0x10802423) DASM(10802423)
775650 0x858 D (0x7b202473) DASM(7b202473)
775656 0x85c D (0x7b200073) DASM(7b200073)
775668 0x80000000 M (0x00c0006f) DASM(00c0006f)
775670 0x8000000c M (0x00000297) DASM(00000297)
...
```

Best,
Michael

TianyangL, 2019-01-11:
Ok, thank you. I'll try again.

msfschaffner, 2019-01-28:
Any news on this one? Otherwise I would like to close this issue...

closed 2019-01-28 by zarubaf

TianyangL, 2019-02-12:
OK, Thank you.
Regards,
刘天阳
---
合肥君正科技有限公司
Ingenic Technology CO.,LTD.
 
公司地址：安徽省合肥高新区望江西路800号创新园C3楼9层
Addr: 9th floor, C3 building,No.800 Chuangxin Park, Wangjiang West Road,
Gaoxin District, Hefei, China
P.C.:  230088
Http：//www.ingenic.com.cn
Mobile:18019932996
Tel:(86-0551)68998700-6101
Fax:(86-0551)68998701
新浪微博号：北京君正官微
微信号：北京君正
Ingenic Semiconductor CO., LTD.
Business Tel: 0551-68998700-6101

From: msfschaffner 
Sent: Monday, January 28, 2019 8:16 PM
To: pulp-platform/ariane 
Cc: TianyangL ; Author 
Subject: Re: [pulp-platform/ariane] The CPU is always in debug mode during emulation (#157)

Any news on this one? Otherwise I would like to close this issue...