openhwfoundation/cva6 #164: Project IceStorm Compatibility?

sunnystormy, 2019-01-18:
Is there a way to generate verilog files so that I can trying getting the core up and running on a Project IceStorm compatible FPGA? Please let me know, thanks!

msfschaffner, 2019-01-28:
Hi @sunnystormy,

Unfortunately we have no flow for Project IceStorm in place. Note that project IceStorm currently targets the iCE40 series from Lattice, which likely does not offer enough LUT and memory resources to implement an Ariane core.

Michael

msfschaffner, 2019-01-28:
On a Xilinx Kintex 7, an Ariane core requires around 40K LUTs and 37 RAMB36 and 16 DSP slices (without FPU). I do not know the exact conversion factor to get an estimate in Lattice LUTs, but I am pretty sure that it will not fit onto the iCE40 series FPGAs (the largest device has only around 7k LUTs). 

I am not sure however whether Yosis supports all SystemVerilog constructs that we use in Ariane...

msfschaffner, 2019-02-01:
Hi @sunnystormy,

Any other questions, or can we close this issue?

sunnystormy, 2019-02-01:
Hi @msfschaffner!

I think I’m good. I’m glad to know what resources are required for the core. That will help me find a suitable FPGA for it. Thank you!

closed 2019-02-01 by msfschaffner