openhwfoundation/cva6 #165: User store segfault while accessing UART

Bhagyaraja, 2019-01-21:
Hello,
I'm trying access (read/write) UART by using UART physical address:  "0x10000000". But, I'm getting "User store segfault". And in trace_hart_0000.log, I can see an exception (Load/store page Fault) with tval of the above physical address. 
However, I tried changing load-store unit privilege level manually (hardcoded internally) and accessed DRAM by physical address. In this case, I'm able to access DRAM. 
Could you please tell why I'm unable to access UART in the first case and how to access it?

Bhagyaraja, 2019-01-28:
Hello, 

Thanks for the response. I'm simulating with modelsim (with proxy-kernel). Is there any way to map physical address in the user space? If not, can you please tell me whether there any way to access UART or other peripheral with proxy-kernel? 

My main motive is to add a peripheral(crypto module) to SoC and access it. Right now, we integrated our module similar to UART but, unable to access it. Can you suggest me the best way to do it?

msfschaffner, 2019-01-30:
Hi @Bhagyaraja,

In a standard Linux environment you would either write a kernel module (proper way) or use the `mmap` syscall to map physical I/O into user space (hacky way). However, since you are using the PK, I don't think that these are valid options, since the PK is not a real Linux kernel (in fact, it just forwards syscalls to the host system). 

How about using a bare-metal environment to test your crypto module?

Bhagyaraja, 2019-01-31:
Hello @msfschaffner,

Can you please tell me how to generate bare metal elf using toolchain? And which toolchain to use. (There are couple of toolchains present for riscv)

We tried using toolchain that we were using for pulpissimo. With that we are getting issue "tohost and fromhost information is missing".  

Thanks again for your help.

Regards,
Bhagyaraja Adapa

msfschaffner, 2019-02-01:
Hi @Bhagyaraja,

We provide scripts that build the correctly configured bare-metal toolchain for you. 

First make sure you properly configured the paths within:

```ci/path-setup.sh```

The $RISCV env variable should point to the place where you want to store the toolchain.

Then you can build the toolchain by following the steps outlined in the following bash script:

```ci/travis-ci-emul.sh```

The GCC toolchain takes a while to build (~1h on a recent desktop machine).

Once you have the toolchain ready, you could, for instance, compile bare-metal programs by using one of the RISC-V benchmarks as a template.

If you executed all steps in the script above, they are located in

```cd tmp/riscv-tests/benchmarks/```

There is also a makefile in that folder where you can see the correct flags and linker script settings.

Hope that helps, 
Michael

Bhagyaraja, 2019-02-04:
Hello @msfschaffner,

Thank you so much for your support. We are able to access to peripherals with the above procedure.

Regards,
Bhagyaraja Adapa

msfschaffner, 2019-02-04:
Sure, no worries.

closed 2019-02-04 by msfschaffner

Bhagyaraja, 2019-03-18:
Hello @msfschaffner,

We are able to add new peripheral and access it in the bare metal simulation. However, we can't access the added peripheral in the proxy-kernel simulation. Can you tell us, is there any way to access this in the proxy-kernel? (something similar to mmap in lilnux)

Regards,
Bhagyaraja Adapa