openhwfoundation/cva6 #169: Mounting a larger file system on the FPGA board

janhoogerbrugge, 2019-01-28:
Hi,

I want to run some benchmarks on the FPGA board. How to get these on the FPGA board? If I follow
the procedure on https://github.com/pulp-platform/ariane-sdk, then the first partition is too small
and if I make it larger then it does not fit anymore in memory when it loaded with rootfs.  So, I want
to use the second partition of the sdcard. But how to mount this partition on the board. Is there
something still necessary in the RTL code to access the sdcard or does something in the 
kernel to be configured?

Regards,
Jan

Moschn, 2019-01-28:
How big is your benchmark?

Indeed, it is not possible to mount the second partition of the sd card currently. I think there is a problem with interrupts/the plic. 

However, it should be possible to create a bigger partition. You just have to adapt the bootloader aswell, since only the first 16mb of the partition are copied to RAM. This can be changed in [this line](https://github.com/pulp-platform/ariane/blob/master/fpga/src/bootrom/src/main.c#L11) where the second parameter times 512 is the amount of bytes that are copied (2 * 16384 * 512 = 16mb).

janhoogerbrugge, 2019-01-28:
The benchmark is SPEC2006 (integer only). If I want to run all the benchmarks in one go I need about 1G. If I would run the benchmarks one by one (or a few at a time) then perhaps a few 100 MB is also fine. I will experiment with the line in the source code you referred to and see how far I will get.

Thanks!

janhoogerbrugge, 2019-01-28:
One more question. The command to format the sdcard is:

  % sudo sgdisk --clear --new=1:2048:67583 --new=2 --typecode=1:3000 --typecode=2:8300 /dev/sdb # create a new gpt partition table and two partitions: 1st partition: 32mb (ONIE boot), second partition: rest (Linux root)

So there a 32M partition is created but 16M are loaded by the bootloader. Shouldn't these numbers not be the same or will everything go fine as long as you do not put more data in the partition than the bootloader will load?

Moschn, 2019-01-28:
Yes, you are correct. The sizes should probably be the same. I think the rationale behind only copying 16mb was to save a little bit of time during the boot.

janhoogerbrugge, 2019-02-04:
I replaced
  
  sd_copy(dest, boot->first_lba, size);

by 

  sd_copy(dest, boot->first_lba, boot->last_lba - boot->first_lba + 1)

in the bootloader so that I do not have to re-synthesize every time I change the size of the
boot partition. Now I can change the size of the boot partition however if I make it larger
than 60M (remember that I need a few 100MB) then it hangs in sd_copy. If I try to read
64MB it hangs at the end of the copy operation. I guess that it hangs here but I do not know why:

          while (sd_dummy() != SD_DATA_TOKEN)
            ;

BTW. This is also strange code in sd_copy (i is incremented instead of j):

        for (int j = 0; j < 8; i++)
            sd_dummy();

janhoogerbrugge, 2019-02-04:
First I tried much larger sizes and that did not work. It hangs after copying about 64MB of data. Then I tried 30MB. That went fine. Then 40MB. Still fine. Then 60MB. Still going fine. Then I did 64MB that then it hangs in sd_copy. I did not see the 'done' at the end. The 'while (sd_dummy() != SD_DATA_TOKEN)' is the only place where it could hang if I see it correctly. Other loops are limited in iterations.

On thing that was also strange that although it boots fine with a 60MB partition without a lot of data in it, it crashes during booting when I put about 40MB in it. This is what I get:

[   35.530000] Kernel panic - not syncing: write error
[   35.530000] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 4.20.0-rc2 #2
[   35.530000] Call Trace:
[   35.530000] [<ffffffe002949de6>] walk_stackframe+0x0/0xa0
[   35.530000] [<ffffffe002949fe2>] show_stack+0x2a/0x34
[   35.530000] [<ffffffe002c2a7fc>] dump_stack+0x62/0x7c
[   35.530000] [<ffffffe00294e028>] panic+0xe8/0x200
[   35.530000] [<ffffffe00000206c>] populate_rootfs+0x36/0xb2
[   35.530000] [<ffffffe002948060>] do_one_initcall+0x48/0x128
[   35.530000] [<ffffffe000000af4>] kernel_init_freeable+0x12e/0x1ca
[   35.530000] [<ffffffe002c3eab6>] kernel_init+0x12/0xf0
[   35.530000] [<ffffffe002949054>] ret_from_exception+0x0/0xc

janhoogerbrugge, 2019-02-05:
I did some more debugging. If I prepare a 128MB partition that gets loaded with sd_copy then it breaks out of the outer do/while loop due to a crc mismatch (crc != crc_exp). The strange thing is that it reaches the end of sd_copy but it seems that it does not return properly to gpt_find_boot_partition. Is the return address on the stack clobbered?

When I disable the crc check then it stays in the outer do/while loop and hangs on 

  while (sd_dummy() != SD_DATA_TOKEN)
            ;

Any ideas?

Moschn, 2019-02-06:
I have no idea what is causing this behavior. I will look into it next week

jrrk, 2019-02-07:
The GenesysII board has 1GByte of RAM. Only 1 of these chips is wired up in the current MIG config, limiting the total to 512M. However the ariane_soc_pkg.sv variable DRAMLength limits this further to 128M for some reason that I am not aware of, likewise the .DTS file would have to be changed to allow Linux to use that memory. In addition the default IOMMU steals 64M of memory by default leaving you with a paltry 64M, of which 16M is typically used by the kernel and rootfs. So it is not surprising that you are having problems with big ramdisks. I am testing a fix to some of these issues and will submit a pull request when I know more. The most difficult issue to fix will be gaining access to the second DDR3 chip, as it requires hacking the board pinout in conjunction with the Digilent data sheets.

Notwithstanding this, just making the rootfs bigger and bigger is not the way forward, because if dedicates memory that could be used for buffers and applications to a fixed function. If you care to try out my pull requests for NFS (network filing system), I would be interested in your feedback.

janhoogerbrugge, 2019-02-08:
Thanks for your analysis of the problem!

Hopefully you are able to enable the memory size first to 512MB and then to 1GB by enabling the second memory chip. I can imagine that the second step is not so easy.

I can try out your NFS (or other networking stuff) work. Hopefully it is not too complex or time consuming (my time for this will be limited).

jrrk, 2019-02-08:
On further investigation it seems that the two DDR3 chips are connected in parallel in the schematic, so the full memory address space should already be accessible, once other gating factors have been addressed.

closed 2019-02-19 by zarubaf

janhoogerbrugge, 2019-02-20:
Hi,

The clobbered return address that is reported above is coming from this in startup.S from the bootrom:

`  li sp, 0x84000000`

If I replace it by

`  li sp, 0xbffffff0`

then the stack pointer starts at the top of the 1GB. Then I can load the rootfs with more data. Booting is slow as you can imagine. After that I am able to run a few SPEC benchmarks which is also slow because the data cache is not working.

Jan.

Iripi97, 2020-01-30:
@janhoogerbrugge I know it has been a while since you commented this, but how did you run the SPEC benchmarks? Were you using spike + pk? If so, how were you able to get those tools onto Ariane and running properly? Any response would be greatly appreciated.

janhoogerbrugge, 2020-01-31:
Hi,

I used Ariane to validate an architectural concept. I stopped using Ariane
because it switched to a write-through dcache, where my concept needed a
write-back dcache, and because the file system on the FPGA system was
ramdisk based which was too small for the benchmarks. So in the end I used
Spike to validate the concept. That was a pity because I liked Ariane and
its FPGA system.

Jan.

Iripi97, 2020-01-31:
@janhoogerbrugge I am sorry to hear that the cache did not fit the specifications that your project required. I was wondering, how were you able to run Spike on the FPGA? I have tried building the tool and adding it to the second partition of the SD card and it fails to run on the FPGA (pretty sure it is due to it being dynamically linked). Did you follow and specific steps to enable spike to run on your FPGA setup?

jrrk2, 2020-01-31:
It looks like there is some confusion here. Spike was never designed to run on FPGA. If it did, it would only emulate very slowly what the FPGA does at 50MHz or more. However I see no reason why a Rocket based FPGA could not run SPEC benchmarks. You would probably want to top and tail your test programs with performance counter measurements.

Iripi97, 2020-01-31:
@jrrk @janhoogerbrugge I apologize, I misunderstood how to run the SPEC CPU2006 benchmarks. I now know you do not need to use a tool like Spike to run them, I now understand this would just simulate how the benchmarks would operate. Thank you for the responses.

davidschrammel, 2021-03-01:
Hi,
I am wondering why this Issue was closed. I don't think the underlying issue is fixed and also the current 16MB limit is not documented.

In my setup I adapted the ariane-sdk Makefile to (optionally) dynamically change the partition size based on the linux image. I also use a similar patch to the one above for copying the larger partition:
```diff
- res = sd_copy(dest, boot->first_lba, size);
+ res = sd_copy(dest, boot->first_lba, boot->last_lba - boot->first_lba + 1)
```
This should be changed in `fpga/src/bootrom/src/gpt.c` as well as in `openpiton/bootrom/linux/src/gpt.c`.

Is it desirable to have pull request with these changes?
Alternatively I would propose adding a fail-safe to the ariane-sdk Makefile that warns the user when the linux image is too big to prevent unexpected crashes or hangs during boot.
Should I create a separate issue/pr for this?

Moschn, 2021-03-01:
Sure, a PR is very welcome