openhwfoundation/cva6 #173: CACHE_START_ADDR

suppamax, 2019-02-03:
In my simulations I have problems when I set CACHE_START_ADDR = 64'h8000_0000_0000_0000.

I see some data requests are directed to data_if port and some to the bypass_if port (my impression is that the core is toggling which port to use), even if no address falls in the cached region.

I am using ariane version bsc-3-10-18: has this problem been fixed in v4.0.0.0?

suppamax, 2019-02-03:
I have also noticed that this problem occurs for every value >= 0x0100000000000000 (I tested only the values which are power of 2).

msfschaffner, 2019-02-03:
Hi @suppamax, 

I strongly suggest to move to a newer version of Ariane, since we introduced several improvements and bugfixes in the mean time. 

The current master branch should be safe to check out and work with.

Let me know if the problem still persists after upgrading, and I'll take a closer look at the issue.

Best,
Michael

janhoogerbrugge, 2019-02-04:
One (little) thing that you will encounter after updating is that the framing_top module is missing in the code. It needs this even when InclEthernet is set to zero.

What is the status of the Ethernet interface? Any idea when it would be ready? It would be very useful for me.

zarubaf, 2019-02-05:
@jrrk Can probably comment more on the ethernet interface. I thought `framing_top` should be there. You might need to do another:
```
git submodule update --init --recursive
```

jrrk, 2019-02-05:
We took the decision to make the Ethernet peripheral a separate sub-module when it was adapted for open-source implementation. However it is based on an MIT license which is known to be compatible. We have seen some useful operation at 100BaseT speeds (DHCP, ARP, and PING) and I hope to merge 1000BaseT support shortly. I don't know when TCP will be supported as it is a mystery why it is not working, perhaps there is some packet filtering in the latest kernel by default for security reasons. Check the fpga/src/ariane-ethernet submodule to see if it is present. There are same changes to scripts also which you can discover with a fresh checkout.

closed 2019-02-10 by zarubaf