openhwfoundation/cva6 #179: Weird gdb behaviour

jrrk, 2019-02-14:
When the gdb/openocd debugger stops at a breakpoint, register A0 is always zero. If the programmer debugs any expression involving A0 at that point, the debugger may perform an illegal access. Since A0 is the first function parameter, this is a highly likely scenario if it contains a pointer. A workaround of adding a dummy first parameter in the source code to each function that might be stopped in the debugger fixes the problem. Single stepping involving instructions that read A0 gives the expected (non zero) answer. This is suggestive to me that reporting temporary registers used in the debugger service ROM is incorrect, or perhaps an API incompatibility in openocd.

Open On-Chip Debugger 0.10.0+dev-00778-g425c0a7 (2019-01-15-17:51)
GNU gdb (GDB) 8.0.50.20170724-git
ariane development branch (based on master branch)

msfschaffner, 2019-02-14:
Hi Jonathan, I saw the same behavior, and I think it is linked to the patch that we implemented to make the debug module offsettable to arbitrary addresses. For that we had to add a second dscratch register that saves the state of A0. We should check whether a hardware fix for this erroneous behavior is possible such that we do not have to patch openocd. I can take a look at this tomorrow.

msfschaffner, 2019-02-14:
Hi Jonathan, I am unfortunately travelling at the moment - could you check whether it works on the FPGA with this patch?

closed 2019-02-14 by zarubaf

jrrk, 2019-02-14:
OK, I set it going, thanks for your prompt attention. I suppose it must work because Florian has closed the issue!

zarubaf, 2019-02-14:
Not on purpose :) I think the merge closed the issue. Feel free to re-open it in case it doesn't work for you.

jrrk, 2019-02-14:
I get the following error:

Starting DRC Task
INFO: [DRC 23-27] Running DRC with 8 threads
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_ariane_peripherals/i_axi2apb_64_32_plic/slave_w_buffer_i/i_axi_single_slice/i_fifo/impl/i_fifo_v3/target_irq_completes_id[1]_1[0] has multiple drivers: i_ariane_peripherals/i_axi2apb_64_32_plic/slave_ar_buffer_i/i_axi_single_slice/i_fifo/impl/i_fifo_v3/genblk1[0].complete_array[2][0]_i_3/O, and i_ariane_peripherals/i_axi2apb_64_32_plic/slave_w_buffer_i/i_axi_single_slice/i_fifo/impl/i_fifo_v3/genblk1[0].complete_array[2][0]_i_2/O.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_ariane_peripherals/i_axi2apb_64_32_plic/slave_w_buffer_i/i_axi_single_slice/i_fifo/impl/i_fifo_v3/target_irq_completes_id[1]_1[1] has multiple drivers: i_ariane_peripherals/i_axi2apb_64_32_plic/slave_ar_buffer_i/i_axi_single_slice/i_fifo/impl/i_fifo_v3/genblk1[0].complete_array[2][0]_i_7/O, and i_ariane_peripherals/i_axi2apb_64_32_plic/slave_w_buffer_i/i_axi_single_slice/i_fifo/impl/i_fifo_v3/genblk1[0].complete_array[2][0]_i_6/O.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_ariane_peripherals/i_axi2apb_64_32_plic/slave_w_buffer_i/i_axi_single_slice/i_fifo/impl/i_fifo_v3/target_irq_completes_id[1]_1[2] has multiple drivers: i_ariane_peripherals/i_axi2apb_64_32_plic/slave_ar_buffer_i/i_axi_single_slice/i_fifo/impl/i_fifo_v3/genblk1[0].complete_array[2][0]_i_4/O, and i_ariane_peripherals/i_axi2apb_64_32_plic/slave_w_buffer_i/i_axi_single_slice/i_fifo/impl/i_fifo_v3/genblk1[0].complete_array[2][0]_i_5/O.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_ariane_peripherals/i_plic/plic_interface_instance/D[0] has multiple drivers: i_ariane_peripherals/i_plic/plic_interface_instance/id_of_largest_priority_q[1][0]_i_1/O, and i_ariane_peripherals/i_plic/plic_interface_instance/id_of_largest_priority_q[1][0]_i_2/O.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_ariane_peripherals/i_plic/plic_interface_instance/D[1] has multiple drivers: i_ariane_peripherals/i_plic/plic_interface_instance/id_of_largest_priority_q[1][1]_i_1/O, and i_ariane_peripherals/i_plic/plic_interface_instance/id_of_largest_priority_q[1][1]_i_2/O.
INFO: [Project 1-461] DRC finished with 5 Errors
INFO: [Project 1-462] Please refer to the DRC report (report_drc) for more information.
ERROR: [Vivado_Tcl 4-78] Error(s) found during DRC. Opt_design not run.

This is a fresh checkout of 426b73e56b5c693bddd6c9e54922e9f5dfa3eaa4 from 	https://github.com/msfschaffner/ariane 

I could stay and debug it, but bearing in mind it's Valentine's day I do not think my wife would like it ...

msfschaffner, 2019-02-14:
Ahh that commit contains another unrelated "fix" in the PLIC, which still produces some driver violations. I suggest you do a fresh checkout of `f5bc68618db70beee5168dfbdf40e8a7141c7c4d`  from our master https://github.com/pulp-platform/ariane/commits/master

jrrk, 2019-02-15:
That version builds, and is certainly behaving as I would expect on cursory examination with the debugger (a0 gets values loaded into it), and the source level debugging prints plausible values now. Thanks again for the prompt response.

commit f5bc6861 (2019-02-14): Fix register access issue in debug mode, only affects A0 (fix #179). (#180)
commit 1b233c12 (2019-02-20): Maturity fixes and AXI extensions for write-through cache system (#188)

* : Fix PITON_ARIANE define issues
* Fix write-back / cache read collision issue in serpent dcache.
* Add separate bootrom / device tree for openpiton (hardcoded for 1x1 tile config at the moment).
* Bootrom generation update (better compatibility with older python versions), new bootrom for OpenPiton+Ariane.
* Fix assertion in icache.
* Correct JTAG timing constraints.
* Fix parameter type in fpga toplevel (fix #168).
* Remove conflicting bootrom from fpga file list.
* This flushs the branch predictors when entering exce [... cut]
commit 07df1426 (2019-03-18): Maturity fixes and AXI extensions for write-through cache system (#188)

* : Fix PITON_ARIANE define issues
* Fix write-back / cache read collision issue in serpent dcache.
* Add separate bootrom / device tree for openpiton (hardcoded for 1x1 tile config at the moment).
* Bootrom generation update (better compatibility with older python versions), new bootrom for OpenPiton+Ariane.
* Fix assertion in icache.
* Correct JTAG timing constraints.
* Fix parameter type in fpga toplevel (fix #168).
* Remove conflicting bootrom from fpga file list.
* This flushs the branch predictors when entering exce [... cut]