openhwfoundation/cva6 #181: make torture-gen: java.lang.NullPointerException

suppamax, 2019-02-15:
Hi, 

I get this error when I do "make torture-gen".
That's what I have installed on my machine
openjdk version "10.0.2" 2018-07-17
OpenJDK Runtime Environment (build 10.0.2+13-Ubuntu-1ubuntu0.18.04.4)
OpenJDK 64-Bit Server VM (build 10.0.2+13-Ubuntu-1ubuntu0.18.04.4, mixed mode)

Any suggestion?
Thanks

cd tmp/riscv-torture && java -jar sbt-launch.jar 'generator/run'
[info] Loading project definition from /home/masgia01/Designs/pulp/ariane/tmp/riscv-torture/project
java.lang.NullPointerException
	at java.base/java.util.regex.Matcher.getTextLength(Matcher.java:1769)
	at java.base/java.util.regex.Matcher.reset(Matcher.java:416)
	at java.base/java.util.regex.Matcher.<init>(Matcher.java:253)
	at java.base/java.util.regex.Pattern.matcher(Pattern.java:1130)
	at java.base/java.util.regex.Pattern.split(Pattern.java:1249)
	at java.base/java.util.regex.Pattern.split(Pattern.java:1322)
	at sbt.IO$.pathSplit(IO.scala:744)
	at sbt.IO$.parseClasspath(IO.scala:859)
	at sbt.compiler.CompilerArguments.extClasspath(CompilerArguments.scala:62)
	at sbt.compiler.MixedAnalyzingCompiler$.withBootclasspath(MixedAnalyzingCompiler.scala:189)
	at sbt.compiler.MixedAnalyzingCompiler$.searchClasspathAndLookup(MixedAnalyzingCompiler.scala:167)
	at sbt.compiler.MixedAnalyzingCompiler$.apply(MixedAnalyzingCompiler.scala:177)
	at sbt.compiler.IC$.incrementalCompile(IncrementalCompiler.scala:138)
	at sbt.Compiler$.compile(Compiler.scala:128)
	at sbt.Compiler$.compile(Compiler.scala:114)
	at sbt.Defaults$.sbt$Defaults$$compileIncrementalTaskImpl(Defaults.scala:814)
	at sbt.Defaults$$anonfun$compileIncrementalTask$1.apply(Defaults.scala:805)
	at sbt.Defaults$$anonfun$compileIncrementalTask$1.apply(Defaults.scala:803)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:226)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:226)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:235)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:226)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:226)
	at sbt.ConcurrentRestrictions$$anon$4$$anonfun$1.apply(ConcurrentRestrictions.scala:159)
	at sbt.CompletionService$$anon$2.call(CompletionService.scala:28)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:514)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1135)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:844)
[error] (compile:compileIncremental) java.lang.NullPointerException
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? i
[warn] Ignoring load failure: no project loaded.
[error] Expected ID character
[error] Not a valid command: generator
[error] generator/run
[error]          ^
Makefile:358: recipe for target 'torture-gen' failed
make: *** [torture-gen] Error 1

zarubaf, 2019-03-27:
I don't have a clue about this. But as this is related to `torture` you might want to open an issue on the `torture` repo.

closed 2019-03-27 by zarubaf

suppamax, 2019-03-27:
FYI: the problem was related to the java version.

zarubaf, 2019-03-27:
Thanks for clarifying it. It might help others in the future!