openhwfoundation/cva6 #182: PLIC Enhancements

zarubaf, 2019-02-15:
The PLIC needs some address map fixes.

jrrk, 2019-03-27:
irq_enables_o[] has a first dimension of 1 by default, but is accessed by indexes up to 31. Should the entire address expression be modulo (%) NUM_GATEWAYS ?

zarubaf, 2019-03-27:
I hope that the PLIC will leave this repository soon for a better implementation.

jrrk, 2019-03-27:
I agree, just looking for a quick hack to try out my interrupt driven driver while waiting for the new PLIC to come along.

zarubaf, 2019-04-03:
https://github.com/lowRISC/rv_plic/pull/1#issuecomment-479400990

What problems do you see with the new `plic` vs the old `plic`? I never got interrupts greater than `0` to work. So I am already positively surprised by the amount of interrupts actually arriving.

zarubaf, 2019-04-03:
> It would be even better if Ariane had a coherent AXI DMA slave port
Is there some Linux specification to what interface the OS would expect (e.g. reusing existing drivers)? For me, that sounds like something which could be done in a student project.

jrrk, 2019-04-03:
Most of the Linux drivers use DMA. It tends to be low-end solutions that do not. I would suggest your hypothetical student prototypes the functionality in spike and then implements the hardware to suit. I already have this prototype available: https://github.com/jrrk/riscv-isa-sim/tree/ariane

closed 2019-04-19 by zarubaf