openhwfoundation/cva6 #185: Follow up on DDR Linux Reporting Issue

zarubaf, 2019-02-15:


janhoogerbrugge, 2019-02-18:
Hi,

You need to run 'make all' in fpga/src/bootrom. This will update ariane.dts and put it in bootroom.sv.

This is what it was:

```
# cat /proc/meminfo | grep MemTotal
MemTotal:          56092 kB
```

And this is what it becomes:

```
# cat /proc/meminfo | grep MemTotal
MemTotal:         959708 kB
```

Jan

zarubaf, 2019-02-18:
That is nice! Do you also see the memory reported during Linux boot? I will try rebuilding the bitstreams in a minute.

janhoogerbrugge, 2019-02-18:
This is what I see during boot:

[    0.000000] Memory: 954308K/1046528K available (3050K kernel code, 188K rwdata, 939K rodata, 5403K init, 753K bss, 92220K reserved, 0K cma-reserved)

So that looks good.

zarubaf, 2019-02-18:
Perfect and thanks, then I am going to close this issue!

closed 2019-02-18 by zarubaf