openhwfoundation/cva6 #200: Problem accessing new peripheral in proxy-kernel simulation. Continuation of #165

Bhagyaraja, 2019-03-18:
Hello @msfschaffner,

We are able to add new peripheral and access it in the bare metal simulation(following #165 steps). However, we are unable to access the added peripheral in the proxy-kernel simulation. Can you please tell us, is there any way to access this in the proxy-kernel? (something similar to mmap in Linux)

Regards,
Bhagyaraja Adapa

jrrk, 2019-03-20:
When you use mmap() in Linux, it prepares a mapping using the memory management unit. However in bare-metal mode (by which I assume you mean machine mode, but can also mean operating system free mode), the MMU is always off so you should have access to the whole of memory without any protection, in which case you can access your peripheral by just casting the address of the peripheral to a suitable sized pointer. Apologies in advance if this is not the question you are asking.

msfschaffner, 2019-03-20:
@jrrk I think this is not possible in this case since they are trying to run in user space on the RISC-V proxy-kernel (which throws a "user store segfault" as described in this issue: #165). 

@Bhagyaraja I do not really have much experience with the proxy-kernel TBH. It could be that you have to adapt the proxy kernel for your purpose in order to provide an equivalent mechanism to mmap() or a device driver (i.e., you need to add a mechanism that can perform your access to I/O memory from machine mode).

Bhagyaraja, 2019-03-21:
Thanks for your response. 

Proxy kernel guys also responded ([Link for the response](https://github.com/riscv/riscv-pk/issues/149)) in similar lines and we can't use the proxy kernel to access new peripherals. 

Regards,
Bhagayraja Adapa

closed 2019-03-21 by Bhagyaraja