openhwfoundation/cva6 #204: problem run pk on ariane

monrick0303, 2019-03-21:
hello,
I want load and run simple program through pk on Ariane, but has some problems on running pk.
Follow are my steps:
1) pk -> pk.bin <riscv64-unknown-elf-objcopy -S -O binary --change-addresses -0x80000000 pk pk.bin>
2) parting sdcard <sudo sgdisk --clear --new=1:2048:67583 --new=2 --typecode=1:3000 --typecode=2:8300 /dev/sdc>
3) write pk.bin into sdcard partition 1 <sudo dd if=pk.bin of=/dev/sdc1 status=progress oflag=sync bs=1M>
4) insert sdcard into genesys2 and turn on power
After "copying boot ....... image done", pk get stuck.
I use printm to debug, and found pk stuck at mret in enter_supervisor_mode function.
Does anyone have same problem or know why stuck?
In addition, I found the value of arg0(a0) is not same between ariane and spike.
What are arg0(a0) and arg1(a1) be used for?
thanks

zarubaf, 2019-03-22:
I am not sure it makes a lot of sense to run `pk` without an actual payload. `pk` will look to load an `elf` file through system calls which it proxies via the debug module to a host computer. Unless you provisioned that you will not have success in running programs. `a0` and `a1` contain are the argument registers and contain the `coreid` and the address of the device tree respectively. As the bootloader is different to `spike` at least `a1` should have a different value.

monrick0303, 2019-03-22:
Does the payload mean the elf that I want to load and run?
If yes, how can I provision an actual payload for PK? 

Thanks

zarubaf, 2019-03-22:
Yes. It’s not going to be trivial to provision the payload. Maybe you want to use bbl and include your program as a payload.

monrick0303, 2019-03-24:
I had tried to use bbl and include my hello program as a payload by follow steps.
  1) <riscv64-unknown-elf-gcc -o hello hello.c>
  2) reconfig <../riscv-pk/configure --PREFIX=$RISCV --HOST=riscv64-unknown-elf CC=riscv64-unknown-elf-gcc OBJDUMP=riscv64-unknown-elf-objdump --with-payload=hello>
  3) <make>
  4) bbl -> bbl.bin <riscv64-unknown-elf-objcopy -S -O BINARY --CHANGE-ADDRESSES -0X80000000 bbl bbl.bin>
  5) parting sdcard <sudo sgdisk --CLEAR --NEW=1:2048:67583 --NEW=2 --TYPECODE=1:3000 --TYPECODE=2:8300 /dev/sdc>
  6) write bbl.bin into sdcard partition 1 <SUDO DD IF=BBL.BIN OF=/DEV/SDC1 STATUS=PROGRESS OFLAG=SYNC BS=1M>
  7) insert sdcard into genesys2 and turn on power
But it also get stuck like pk. Does my step have some wrong?
In addition,I had traced pk source code and found pk will print <please tell me elf> if not find elf file.
But when I run pk on ariane without elf, pk don't print <please tell me elf>, Is that normal?

monrick0303, 2019-03-27:
Briefly, how to build a payload that include my program?

Thanks.

zarubaf, 2019-03-27:
Packaging it with `bbl` should work. (We are doing the same for the Linux payload). But obviously, you can't expect it to run arbitrary glibc functions (as you don't have an OS taking care of it). You'll need to call into SBI to print to UART. If you want to run a user-program it might be the easiest to just package it on the `rootfs` overlay in `ariane-sdk` and start it in Linux.

Because `pk` is communicating with the host to get the `elf`. But on the FPGA you run untethered and hence it can't find the `elf` file. Either you provide means of tethering your system to the host (e.g. via semi-hosting over GDB) or other means.

zarubaf, 2019-04-16:
Closing this due inactivity. Feel free to re-open if needed.

closed 2019-04-16 by zarubaf