openhwfoundation/cva6 #205: Confused about caches.

hyf6661669, 2019-03-22:
There are two kinds of caches in ariane, "standard cache" and "OpenPiton cache". Can you tell me some information about the difference between them? For example, the difference in consumed LUTs/FFs, max operating frequency, hit accuracy, etc. 

Much thanks!

zarubaf, 2019-03-22:
OpenPiton Cache has been renamed to write-through Cache. This also captures the main difference as the std cache is write back. Max frequency should be similar. Hit accuracy depends on your configuration and is the same for both caches. Unfortunate, I can’t give you estimates on area yet.

hyf6661669, 2019-03-23:
@zarubaf I find that you use a 8-bit LFSR to generate random index for cache replacement, can you tell me why don't you use PLRU for cache replacement?

zarubaf, 2019-03-27:
No particular reason, except that the LFSR is pretty straight forward to implement. It should be fairly easy to change that actually and might be an interesting thing to try out. We already have a `PLRU` tree for our TLBs 
https://github.com/pulp-platform/ariane/blob/20e887cd2bb0c0e2609c84e38ab2ad17e5a37762/src/tlb.sv#L128-L202 So it shouldn't be too hard to replace the LFSR with that.

hyf6661669, 2019-03-27:
@zarubaf  
Have you tested the performance gap between "random selection" and PLRU?

zarubaf, 2019-03-27:
Not yet.

raulbehl, 2019-03-27:
> @zarubaf I find that you use a 8-bit LFSR to generate random index for cache replacement, can you tell me why don't you use PLRU for cache replacement?

Hi @hyf6661669, 
After reading your comment I got a bit confused on the usage of LFSR to generate random **index** for cache replacement. That is what prompted me to check the `miss_handler.sv` file and realised that it was indeed used for generating the random cache **way** for replacement. 

Perhaps this is what you also meant and I think this could've been a small typo. It would be great if you could update your comment as it would readers a better understanding?

Thank you,
Rahul

hyf6661669, 2019-03-27:
@raulbehl 
Hi, you can have a look at ```ariane/src/cache_subsystem/std_icache.sv'''. There is a instantiated module:
```
    lfsr_8bit #(.WIDTH (ICACHE_SET_ASSOC)) i_lfsr (
        .clk_i          ( clk_i       ),
        .rst_ni         ( rst_ni      ),
        .en_i           ( update_lfsr ),
        .refill_way_oh  ( random_way  ),
        .refill_way_bin (             ) // left open
    );
```
I find that this module is used for icache replacement. But I haven't read codes about dcache. Therefore, I don't know the replacement policy for dcache.

msfschaffner, 2019-03-28:
Hi @hyf6661669, 

We use pseudo-random replacement in all of our L1 cache systems (data and instruction caches, write-back and write-through flavour). This replacement strategy is simple to implement and does not require an additional array with `log2(NUM_WAYS)` bits per set. It is also sometimes argued that pseudo random is better at breaking cyclic dependency patterns than other replacement strategies. 

Note that this is not an uncommon design choice, as it allows for simple and fast L1 implementations. ARM A9 also implements this replacement strategy in both the L1 data and instruction caches, [see](http://infocenter.arm.com/help/topic/com.arm.doc.100511_0401_10_en/arm_cortexa9_trm_100511_0401_10_en.pdf) on page 118, Section 7.11.

For higher level caches such as L2 and L3, PLRU or similar is likely a better choice.

hyf6661669, 2019-03-28:
@msfschaffner 
Thanks for your explanation.

closed 2019-03-29 by hyf6661669