openhwfoundation/cva6 #207: clint/axi_lite_interface.address_o

suppamax, 2019-03-26:
I think ariane/src/clint/axi_lite_interface.sv line 93 might be dangerous.
In my understanding a read request to the clint will not be started until the next cycle, when en_o is set.
Moreover, it might create the path axi_req_i.aw_valid-address_o-rdata-axi_resp_o.r.data, which might be bad for timing closure.

zarubaf, 2019-03-26:
Yes, `en_o` is registered. Are you experiencing problems closing timing on this path? I haven't seen getting this path critical in the last three chips. In any case, it should be easy to fix.

suppamax, 2019-03-26:
In my case I have Ariane (version of October 2018) and the Clint connected through an AXI interconnect.
The critical path looks like this

ariane_i/i_std_cache_subsystem/i_nbdcache/i_miss_handler/i_miss_axi_adapter/axi.aw_addr
axi_interconnect_i/axi_node_i/_RESP_BLOCK_GEN[1].RESP_BLOCK/awaddr_i
axi_interconnect_i/axi_node_i/_REQ_BLOCK_GEN[8].REQ_BLOCK/awvalid_i
axi_interconnect_i/axi_node_i/_REQ_BLOCK_GEN[8].REQ_BLOCK/AW_ALLOCATOR/AW_ARB_TREE/data_req_o
axi_interconnect_i/axi_node_i/_REQ_BLOCK_GEN[8].REQ_BLOCK/AW_ALLOCATOR/awvalid_o
clint_i/axi_lite_interface_i/slave.aw_valid
clint_i/axi_lite_interface_i/address_o
clint_i/rdata
axi_interconnect_i/axi_node_i/master_rdata_i
axi_interconnect_i/axi_node_i/_RESP_BLOCK_GEN[1].RESP_BLOCK/rdata_i
...
ariane_i/ex_stage_i/lsu_i/i_lsu_arbiter/i_ld_fifo/mem_q_reg

zarubaf, 2019-03-27:
Thanks. See #208

closed 2019-03-27 by zarubaf