openhwfoundation/cva6 #210: Issues with Virtual L1 cache

kratiagrawal, 2019-03-27:
Hi, I'm trying to implement virtual cache (L1) in place of the physical cache. There is a check on tag which decides whether it should bypass the cache or not (CACHE_START_ADDR), but this address corresponds to the physical tag. How can I have an equivalent condition for virtual tag?

msfschaffner, 2019-03-28:
Hi @kratiagrawal,

If you do not have access to the physical address, you would have to keep track whether a certain memory page is cached or not (but you usually cannot infer that from looking at the virtual address only).

Which cache do you intend to replace and what kind of cache configuration do you want to implement exactly?

Our L1 instruction and data caches are both "virtually indexed, physically tagged" in order to accommodate virtual-to-physical address translation efficiently. 

Note that you will need a virtual-to-physical address translation for such a system to work at some point, and in Ariane this translation is tightly integrated together with the L1 cache system.

kratiagrawal, 2019-03-28:
I am trying to make L1 data cache as virtual (virtual index, virtual tag). On a miss (or bypass), I go to memory with the physical address. Right now, I don't care when the address translation happens, but I want to use it only after an L1 miss.

Could you give me an example of the use case of this check on physical tag that decides whether the address should bypass cache or not? And how would I do that with the virtual address?

It would be great if you could give me some pointers on how to decouple the L1 cache subsystem from the address translation.

Thank you.

zarubaf, 2019-03-28:
In general, you cannot decide, based on the virtual address, whether to bypass the cache or not - you need the physical address for this. Furthermore, you will need to add measures to check on potential aliasing even when you hit on the cache.

kratiagrawal, 2019-03-29:
Yes, I have plans to address the aliasing issue. For the cache bypassing, I understand there is no straight forward way to do it with virtual address but there should be some other means to do it. It would be great if you could explain any ideas that you might have in the context of Ariane.

msfschaffner, 2019-03-29:
In our Ariane system there are basically 3 cases where you need to bypass the cache:

1) the cache is disabled. this is easy for you to recognize even with a vi/vt cache.

2) depending on the cache (write-through or write-back) and the employed coherency mechanism, atomic accesses are treated as non-cacheable. but since they come in via a dedicated atomic interface you can easily recognize and separate atomics from other memory operations in your case.

3) You want to make an access into a non-cacheable region (e.g. I/O). In our Ariane system we only have one contiguous cacheable memory region (basically the whole DRAM). Everything else is considered to be non-cacheable and this is also where peripherals like the debug module, bootrom and UART are mapped. Accesses to those regions *must* not be cached due to the side effects they may have (e.g., in case of UART, you are reading from / writing to the same registers / FIFOs over and over again). 

Correct me if I am wrong, but afaik, it is in general not possible to detect whether a virtual address maps to a non-cacheable physical address  by just looking at the virtual address itself - you need more information than that from the MMU and this is also why L1 caches are often implemented as vi/pt caches, where the virtual memory translation is performed in parallel to the tag lookup. If you do not have a mechanism to translate at least parts of your virtual address before/while going into L1, this information would have to be encoded into the virtual address itself, and that would need special OS support.

Could you maybe explain why you would like to go for a vi/vt cache in your case such that we can better understand what you are trying to achieve?

zarubaf, 2019-04-16:
Closing this due inactivity. Feel free to re-open if needed.

closed 2019-04-16 by zarubaf