openhwfoundation/cva6 #211: Variane_testharness seems to get stuck

Daryan7, 2019-04-03:
Hi,

I'm new to the riscv tool chain and Ariane. I have installed the tools from [https://github.com/riscv/riscv-tools](url)  and downloaded the master branch of Ariane, following the getting started guide. I managed to install everything, but when I try to execute Variane_testharness with pk and a simple hello world example, the program never ends. I run the same command as in the guide:

`work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf
`

But what I see in the trace is that it seems to get stuck with the following instruction:

`1368128 0x8000255a M (0x0000a001) c.j     pc + 0
`

Do you know a possible cause for this problem? Is this normal? I run the program for more than 10 minutes and I don't think it should take that long. Running the same elf file with Spike finishes correctly.

Thanks!

zarubaf, 2019-04-11:
Do you think you could share the source with me? Then I can help to localize it. Sorry for the long radio silence btw.

Daryan7, 2019-04-11:
I am not sure what source are you talking about, sorry. Do you mean the trace file with all the instructions?  the source code I used? or something else?  I only used the hello world example in the getting started guide.

zarubaf, 2019-04-11:
Okay, sorry I misunderstood you. I somehow missed the fact that you trying the hello world which I was suggesting in the readme.

First of all, thank you for pointing this out. I could reproduce the problem on my side. The good news, it is not something related to the core. The problem is that the proxy kernel `pk` tries to read the device tree which is in the bootrom. By convention, the `a1` register points to the base of the device tree. This register is set by the zero stage bootloader (ZSBL). Unfortunately the debug module pulls away the core before the ZSBL is beeing exceuted and points the core to the base of the DRAM. The result is that the register `a1` does not contain the pointer to the device tree but is zeroed. From here `pk` thinks that the memory is zero and aborts.

Unfortunately, right now I don't have an easy fix for this. I'll need to think a bit more about this. Sorry for the inconvenience.

msfschaffner, 2019-04-11:
How about we add a small startup delay to the simDTM by gating the debug req handshake?

zarubaf, 2019-04-16:
Yes, this seems legit. Thanks.

msfschaffner, 2019-05-01:
This has been fixed and merged into the `ariane-dev` branch. @Daryan7 could you check whether this problem still persists on your side and close this issue if not?

jdeharor, 2019-05-07:
Hi,

Sorry for the delay. No, I cloned the ariane-dev branch and the result is different, the bootrom that sets the device tree pointer on the register a1 is doing well (I think), but nothing loads the program, so the core starts fetching from the DRAM before the debug module does anything. Here's the trace:

                 266 0x10000 M (0x0010041b) addiw   s0, zero, 1
                 271 0x10004 M (0x01f41413) slli    s0, s0, 31
                 276 0x10008 M (0xf1402573) csrr    a0, mhartid
                 281 0x1000c M (0x00000597) auipc   a1, 0x0
                 286 0x10010 M (0x07458593) addi    a1, a1, 116
                 291 0x10014 M (0x00008402) c.jr    s0
                 302 0x80000000 M (0xdeadbeef) jal     t4, pc - 0x24a16

Maybe I am doing something wrong, but I followed the same steps as the first time.

Daryan7, 2019-05-07:
BTW, @jdeharor is another account I have. I replied with that one because I didn't look in which one I was logged in. Sorry for the confusion.

> Hi,
>
> Sorry for the delay. No, I cloned the ariane-dev branch and the result is different, the bootrom that sets the device tree pointer on the register a1 is doing well (I think), but nothing loads [...]

zarubaf, 2019-05-13:
Hm, that is strange. It is working for me now.
```

Daryan7, 2019-05-20:
I cloned again the repository with the ariane-dev branch and refhreshed all submodles. Now it seems that the result is the same, the program never finishes. The trace is this one:

                 266 0x10000 M (0x0010041b) addiw   s0, zero, 1
                 271 0x10004 M (0x01f41413) slli    s0, s0, 31
                 276 0x10008 M (0xf1402573) csrr    a0, mhartid
                 281 0x1000c M (0x00000597) auipc   a1, 0x0
                 286 0x10010 M (0x07458593) addi    a1, a1, 116
                 291 0x10014 M (0x00008402) c.jr    s0
                 311 0x10040 M (0xf1402573) csrr    a0, mhartid
                 316 0x10044 M (0x00000597) auipc   a1, 0x0
                 321 0x10048 M (0x03c58593) addi    a1, a1, 60
                 326 0x1004c M (0x10500073) wfi
                 331 0x10050 M (0x0000bff5) c.j     pc - 4
                 336 0x1004c M (0x10500073) wfi
                 341 0x10050 M (0x0000bff5) c.j     pc - 4

Did you use the same branch?
Thanks

msfschaffner, 2019-05-27:
I tried again with the newest `ariane-dev` branch and it works for me. Maybe you did not wait long enough? It takes a while...

jdeharor, 2019-05-31:
I tried again on another computer. You were right, after waiting a little bit longer it finally worked.

Thanks to all for the help!!

msfschaffner, 2019-05-31:
You're welcome!

commit 7232cb18 (2019-04-19): testharness: Add delay to `debug_req` signal (fixes #211)

Upon start the `debug_req` signal is being delayed for a couple of
cycles to give the core enough time to initialize `a0` and `a1` with its
`hartid` and a pointer to the `dts` (e.g. executing the first
instructions of the ZSBL). This fixes issue #211.
commit 6ca41217 (2019-06-04): testharness: Add delay to `debug_req` signal (fixes #211)

Upon start the `debug_req` signal is being delayed for a couple of
cycles to give the core enough time to initialize `a0` and `a1` with its
`hartid` and a pointer to the `dts` (e.g. executing the first
instructions of the ZSBL). This fixes issue #211.
closed 2019-05-31 by msfschaffner